SmingHub/Sming #2: virtualWire library from arduino

alonewolfx2, 2015-03-22:
```
Compatible with standard Arduino libraries - use any popular hardware in few lines of code
```
i saw this line on readme and i am very exciting. how can i use this library? can you ad this library on your project?

anakod, 2015-03-22:
I have plan to add different popular libraries to project, but if you want, you can try it by yourself, because, usually just needed:
1. Check variables types ("int" is int16 on Arduino, but int32 on ESP)
2. Check file includes. If you meet problems, in most cases just comment out problem files.

All other things should work without changes.

alonewolfx2, 2015-03-22:
i tried but it seems some internal things not implemented. soo i am waiting you for adding virtualWire library.
```
Libraries/VirtualWire/VirtualWire.cpp:519:5: error: 'TCCR1A' was not declared in this scope
     TCCR1A = 0; // Output Compare pins disconnected
     ^
Libraries/VirtualWire/VirtualWire.cpp:520:5: error: 'TCCR1B' was not declared in this scope
     TCCR1B = _BV(WGM12); // Turn on CTC mode
     ^
In file included from Wiring/../Wiring/WiringFrameworkDependencies.h:21:0,
                 from Wiring/Arduino.h:7,
                 from Libraries/VirtualWire/VirtualWire.h:223,
                 from Libraries/VirtualWire/VirtualWire.cpp:9:
Libraries/VirtualWire/VirtualWire.cpp:520:18: error: 'WGM12' was not declared in this scope
     TCCR1B = _BV(WGM12); // Turn on CTC mode
                  ^
Wiring/../Wiring/BitManipulations.h:11:28: note: in definition of macro '_BV'
 #define  _BV(bit)   (1 << (bit))
                            ^
Libraries/VirtualWire/VirtualWire.cpp:527:5: error: 'OCR1A' was not declared in this scope
     OCR1A = nticks;
     ^
Libraries/VirtualWire/VirtualWire.cpp:534:5: error: 'TIMSK' was not declared in this scope
     TIMSK |= _BV(OCIE1A);
     ^
In file included from Wiring/../Wiring/WiringFrameworkDependencies.h:21:0,
                 from Wiring/Arduino.h:7,
                 from Libraries/VirtualWire/VirtualWire.h:223,
                 from Libraries/VirtualWire/VirtualWire.cpp:9:
Libraries/VirtualWire/VirtualWire.cpp:534:18: error: 'OCIE1A' was not declared in this scope
     TIMSK |= _BV(OCIE1A);
                  ^
Wiring/../Wiring/BitManipulations.h:11:28: note: in definition of macro '_BV'
 #define  _BV(bit)   (1 << (bit))
                            ^
Libraries/VirtualWire/VirtualWire.cpp: At global scope:
Libraries/VirtualWire/VirtualWire.cpp:785:4: error: expected constructor, destructor, or type conversion before '(' token
 ISR(VW_TIMER_VECTOR)
    ^
Libraries/VirtualWire/VirtualWire.cpp:870:1: error: expected '}' at end of input
 } // extern "C"
 ^
C:/Espressif/examples/Sming/Sming/Makefile:326: recipe for target 'out/build/Libraries/VirtualWire/VirtualWire.o' failed
mingw32-make.exe: *** [out/build/Libraries/VirtualWire/VirtualWire.o] Error 1
```

anakod, 2015-03-22:
I'm google it now and looks like WGM12 is a timer register. In ESP, timer model was different, so I'm think what this will make some problems. Simplest way - try to check different radio library :) What hardware you are trying to connect?

anakod, 2015-03-22:
Other way - you can try change timers to Sming class "Timer". But it can require some non-trivial changes.

alonewolfx2, 2015-03-22:
i am using this cheap modules. firstly i am trying to find other libraries :)
http://www.ebay.com/itm/433Mhz-RF-transmitter-and-receiver-kit-for-Arduino-ARM-WL-MCU-Raspberry-pi-new-/131463741512?pt=LH_DefaultDomain_0&hash=item1e9bd98048

hreintke, 2015-10-03:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-03 by hreintke