SmingHub/Sming #3: system restarting every 15 minutes.

alonewolfx2, 2015-03-23:
it seems system have wdt issue.sming going to  wdt reset every 15 minutes. i tried wdt_feed but its not working. have you any idea?

anakod, 2015-03-23:
Yes, I meet this problem, I think problem in SDK, not in Sming, but not sure.

alonewolfx2, 2015-03-23:
normaly timers overflow every 35 minutes but it seems this value is 15 minutes in sming. we must to clear or disable wdt for stable working.

anakod, 2015-03-23:
Disable WDT not good idea for stable working :)

anakod, 2015-03-23:
I'm sure should be some simple fix for that. May be need to google more SDK WDT api.

alonewolfx2, 2015-03-23:
can you join this chat https://gitter.im/alonewolfx2/Sming

alonewolfx2, 2015-04-06:
it seems reboot problem resolved with this commit. https://github.com/anakod/Sming/commit/8c98f5db2ab637010b933379d382bbcc843062e6

closed 2015-04-06 by alonewolfx2