SmingHub/Sming #10: Can't compile with esp-open-sdk

ystrem, 2015-03-27:
Hi, I tried to compile it with esp-open-sdk and I get this error, can someone with different SDK confirm that error ?
```
ystrem@ubuntu:~/Sming/Basic_Blink$ make
C+ app/application.cpp
In file included from app/application.cpp:1:0:
include/user_config.h:26:40: fatal error: espinc/c_types_compatible.h: No such file or directory
  #include <espinc/c_types_compatible.h>
                                        ^
compilation terminated.
make: *** [out/build/app/application.o] Error 1
```

anakod, 2015-03-27:
You have used original makefile? Is Paths right?

ystrem, 2015-03-27:
I changed only the two lines.
```
# Base directory for the compiler
#c:/Espressif/xtensa-lx106-elf/bin
XTENSA_TOOLS_ROOT ?= /home/$USERNAME/esp-open-sdk/xtensa-lx106-elf/bin

# base directory of the ESP8266 SDK package, absolute
#c:/Espressif/ESP8266_SDK
SDK_BASE	?= /home/$USERNAME/esp-open-sdk/sdk/

# esptool path and port
SDK_TOOLS	?= /home/$USERNAME/esp-open-sdk/utils
ESPTOOL		?= /home/$USERNAME/esp-open-sdk/esptool/esptool.py #$(SDK_TOOLS)/esptool.exe
ESPPORT		?= COM4
COM_SPEED	?= 230400
```

kireevco, 2015-03-28:
I was able to work-around hard coded windows paths to SDK. You will need to set some environment variables and only then run make. Make will modify it's paths according to variables you set:

```bash
# Make sure the following path is pointing to your esp-open-sdk directory:
export ESP_OPEN_SDK="/home/build/esp-open-sdk"
#######
export SDK_BASE="${ESP_OPEN_SDK}/sdk"
export XTENSA_TOOLS_ROOT="${ESP_OPEN_SDK}/xtensa-lx106-elf/bin"
export SDK_TOOLS="${SDK_BASE}/tools"
export ESPTOOL="${ESP_OPEN_SDK}/esptool/esptool.py"
export COMPILE="gcc"
export PATH=${ESP_OPEN_SDK}/xtensa-lx106-elf/bin:$PATH
ln -s ${SDK_TOOLS}/gen_appbin.py ${SDK_TOOLS}/gen_appbin.exe
chmod +rx ${SDK_TOOLS}/gen_appbin.exe
```

kireevco, 2015-04-06:
@ystrem Do you think we can close this issue?

anakod, 2015-04-06:
Somebody try this instruction from zeo-configuration: https://github.com/anakod/Sming/wiki/build-linux ?

ystrem, 2015-04-24:
@anakod I tested it, but without changes in Makefile, it won't work. I made some changes to Makefile to make them multiplatform. I can't get work gen_appbin.py working in Makefile.

anakod, 2015-04-24:
Why it can't work?
Also @kireevco probably can help you

ystrem, 2015-04-24:
@anakod may be you misunderstood me. I can compile the example after change the path's, but then I can't use the gen_appbin.exe to merge the files(or whatever it does). But I'm on good way with it, I will post it tonight.

anakod, 2015-04-24:
It isn't necessary to merge files, you can flash them by one (target: flash)

kireevco, 2015-04-26:
Please refer to https://github.com/anakod/Sming/wiki/Linux-Quickstart & https://github.com/anakod/Sming/wiki/MacOS-Quickstart

@anakod Issue can be closed, to my mind.

closed 2015-04-26 by kireevco