SmingHub/Sming #15: Errors on compiling with standard and latest Windows SDK and Sming ... ?

Werquinx, 2015-04-03:
My Windows SDK (standard unofficial) is working fine on my other projects (cf ESP-MQTT).

I downloaded your Ming examples and imported them but when I compile the simple blink or any other example, I get the following errors :+1: 

15:12:28 **** Build of configuration Default for project Basic_Blink ****
mingw32-make.exe -f C:/Users/Xavier/Documents/GitHub/Sming/Basic_Blink/Makefile all 
C+ ../Sming/system/stringconversion.cpp
In file included from c:/Espressif/ESP8266_SDK/include/user_interface.h:11:0,
                 from ../Sming/system/include/esp_systemapi.h:10,
                 from include/user_config.h:29,
                 from ../Sming/system/stringconversion.cpp:1:
c:/Espressif/ESP8266_SDK/include/lwip/ip_addr.h:213:23: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                       ^
c:/Espressif/ESP8266_SDK/include/lwip/ip_addr.h:213:31: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                               ^
c:/Espressif/ESP8266_SDK/include/lwip/ip_addr.h:213:40: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                                        ^
c:/Espressif/ESP8266_SDK/include/lwip/ip_addr.h:213:49: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                                                 ^
cc1plus.exe: all warnings being treated as errors
C:/Users/Xavier/Documents/GitHub/Sming/Basic_Blink/Makefile:330: recipe for target 'out/build/../Sming/system/stringconversion.o' failed
mingw32-make.exe: *** [out/build/../Sming/system/stringconversion.o] Error 1

15:12:29 Build Finished (took 485ms)

Did I miss something ? Does anything special need to be done to integrate your environment into Windows SDK ?

Thanks.

anakod, 2015-04-03:
> If you have SDK v0.9.5, please rename "C:\Espressif\ESP8266_SDK\include\lwip" to lwip_old

On other side you can just update SDK to last version.

Werquinx, 2015-04-03:
OK Perfect. Now works ;-)

closed 2015-04-03 by Werquinx

aruldd, 2017-03-04:
Facing the same issue in linux (Ubuntu 14.04 x64) using precompiled esp-open-sdk and while compiling basic_blink.

```
Basic_Blink$ make
make[1]: Entering directory `/opt/sming/Sming'
make[2]: Entering directory `/opt/sming/Sming/third-party/esp-open-lwip'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `/opt/sming/Sming/third-party/esp-open-lwip'
make[1]: Leaving directory `/opt/sming/Sming'
C+ app/application.cpp
In file included from /opt/esp-open-sdk/sdk/include/user_interface.h:11:0,
                 from /opt/sming/Sming/system/include/esp_systemapi.h:10,
                 from include/user_config.h:29,
                 from app/application.cpp:1:
/opt/sming/Sming/third-party/esp-open-lwip/include/lwip/ip_addr.h:213:23: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                       ^
/opt/sming/Sming/third-party/esp-open-lwip/include/lwip/ip_addr.h:213:31: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                               ^
/opt/sming/Sming/third-party/esp-open-lwip/include/lwip/ip_addr.h:213:40: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                                        ^
/opt/sming/Sming/third-party/esp-open-lwip/include/lwip/ip_addr.h:213:49: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Werror=literal-suffix]
   LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
                                                 ^
cc1plus: all warnings being treated as errors
make: *** [out/build/app/application.o] Error 1

```

ADiea, 2017-03-04:
Will investigate

ADiea, 2017-03-04:
Please show me how you compile sming lib. The make line

ADiea, 2017-03-04:
```
LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F, 
```
But it should look like (from esp-open-lwip.patch)
```
-  LWIP_DEBUGF(debug, ("%"U16_F".%"U16_F".%"U16_F".%"U16_F,             \
+  LWIP_DEBUGF(debug, ("%" U16_F ".%" U16_F ".%" U16_F ".%" U16_F,             \
```

It looks liek you are not applying the patch, how are you building sming? Delete all files from Sming\third-party\esp-open-lwip\ and then make sming again with ```make rebuild ```

ManchTTE, 2018-02-21:
same problem when compile stringconversion.cpp
please help me.

mingw-get v1.0.2
Information.PlatformVersion='6.3.9600.0'|
Information.PlatformName='Windows 8.1'|
............................
**C+ system/stringconversion.cpp
make[1]: *** [out/build/system/stringconversion.o] Error 1
make[1]: Leaving directory `/C/tools/sming/Sming'
make: *** [reload] Error 2**

[err_cc1plus.txt](https://github.com/SmingHub/Sming/files/1742607/err_cc1plus.txt)

slaff, 2018-02-21:
Can you please compile Sming with the following commands ?
```
cd %SMING_HOME%
make dist-clean
make V=1 
```

ManchTTE, 2018-02-21:
when make dist-clean result of all samples is: no  such host file like.
==================Command Promt================================
make[1]: Entering directory `/C/tools/sming/samples/UdpServer_Echo'
/C/tools/sming/Sming/Makefile-project.mk:95: //C/tools/sming/Sming/Makefile-windows.mk: No such host or network path
make[1]: stat: //C/tools/sming/Sming/Makefile-windows.mk: No such host or network path
make[1]: *** No rule to make target `//C/tools/sming/Sming/Makefile-windows.mk'.  Stop.
make[1]: Leaving directory `/C/tools/sming/samples/UdpServer_Echo'"
=====================================================

slaff, 2018-02-21:
@ManchTTE Are you using the "master" version or the "develop" branch? Can you try these commands then?

```
cd %SMING_HOME%
make clean
make V=1
```

ManchTTE, 2018-02-22:
1. i don't no master or develop but i am install by chocolatey and i resolved 7za extract not respond by manual extract it.
2. when make clean i thing no problem.
please see attachment file.
[makeClean.txt](https://github.com/SmingHub/Sming/files/1746832/makeClean.txt)

ManchTTE, 2018-03-20:
Hi today I can use it normally on another computer.
i thing my Processes[CPU] It may be important.
I have experimented.
1.My computer >Core i5-3470--> when compile stringconversion.cpp are Error cc1plus.exe has stopped.
2.My friend1>Core i5-3470--> when compile stringconversion.cpp > Error cc1plus.exe has stopped.
3.Myfriend2>centrino core i5-24xx when compile stringconversion.cpp > Error cc1plus.exe has stopped.
4.Myfriend3>Core i7-4790 can use it normally with Master Branch.
5.Myfriend4>Core i7-4790 can use it normally with Master Branch.
this data i hobe you know and resolved the problem
thank you staff.

kwis2, 2018-03-20:
I have tested on a lot of machines and found that it is more moon phase dependency than CPU.
I have 3 identical machines with even the same manufacturer batch (I think) and one work from first time, another decided to not to work.
But in serious. In my opinion windows install probelm could be that there is no upadate / upgrade mechanisms an when you install Sming every machine could have different versions of software - Sming installer do not replace existing software. For instance, sometimes I was able to success with install when I installed MinGW into c:/tools/mingw64 directory before Sming installation.