SmingHub/Sming #24: Sming projects don't work separately from sming directory

kireevco, 2015-04-12:
We need to find a way to build sming project (say Blink2) when it is located anywhere outside of Sming directory (say in ~/dev/Blink2 or in c:\User\Blink2).

j0hncc, 2015-04-23:
Is there a way to "upvote" this  :)
:arrow_up:

kireevco, 2015-04-23:
You just did:)

ToSa27, 2015-04-24:
So this might be the reason why i wasn't able to setup a new project using Sming from scratch... I followed the steps provided with the DevKit for Windows and then tried to apply whatever is needed to get it working with sming (makefile adjusted for G++ / linking Sming / etc.)
Once this is fixed, it would be great to have e.g. an entry in the wiki describing the steps to create a new project in eclipse ready for sming development :)

kireevco, 2015-04-27:
All, Please test my branch https://github.com/kireevco/sming/tree/crossplatform
Make sure to set SMING_HOME and ESP_HOME environment variables 
(see corresponding section in https://github.com/anakod/Sming/wiki/Windows-Quickstart & https://github.com/anakod/Sming/wiki/MacOS-Quickstart)

closed 2015-05-12 by anakod