SmingHub/Sming #26: Add ability to write initial web pages to eeprom

alon24, 2015-04-14:
When I create a web server (like Mateo example) I want to be able to also put my initial pages into eeprom so that my code would find it.

Currently you have to burn firmware
connect using ftp
load files 
reboot - not sure about this step

I would like to be able to do this in one step, so that I load my firmware and then with another action (simple no need for loading ftp and such) load my files.
(currently mateo sample connects to known url and downloads the files from there).

kireevco, 2015-04-15:
We could use spiffy to add file system bin to the main firmware. @alonewolfx2 and I did some tests, working somewhat. 
Check this gist out https://gist.github.com/kireevco/dfe1e80fb509d57b0f49 tested on windows and ubuntu.

MrZANE42, 2015-04-22:
Any chance of adding a checksum to the filesystem file table (or similar) and if that isn't ok automatically formatting the filesystem at init?

anakod, 2015-04-23:
> and if that isn't ok automatically formatting the filesystem at init?
At last Sming version it should auto-reseting on firmware flashing.

MrZANE42, 2015-04-23:
My thought was that it should only format if the file table CRC didn't
match the expected CRC. If the CRC/checksum is wrong the 'fat' is wrong
anyway and can't be trusted. This could of course either be done by passing
a parameter when initializing telling it what to do(format or ignore) if
the 'fat' is faulty or by a check fat method  returning true/false. Thereby
making the formating optional.

anakod, 2015-04-23:
Recalculation and updating hash on every file changing cost too much and will reduce flash chip life, I think.

alonewolfx2, 2015-04-23:
@anakod agree

MrZANE42, 2015-04-23:
I didn't mean to suggest it should be done on ever file write/change. Just
when adding/deleting or renaming a file or directory (ie. Making changes
that also changes the 'fat'). The CRC would only be of the fat(struct? )
when saved, not of any of the files them self.

MrZANE42, 2015-04-23:
Hi again.
Just hade a quick look at the spifs filesystem files. Seems there are quite a lot of sanity checks so another CRC might not be needed. the spiffs_mount code in smingframework/services/spiffs/spiffs.c (called from smingframework/appinit/user_main.cpp) doesn't return the mount status/failure codes up to user level. maybe the spiffs_mount code could (in certain cases when getting certain fault codes) do the formating automaticaly. I'll see if I can find some time to try this out.

anakod, 2015-04-23:
Great, please notify us if it will help!

kireevco, 2015-05-01:
Still working on it - need to solve cross-platform factor & dependencies for srec, spiffy, bc. We don't want to add dependencies if possible.

kireevco, 2015-05-11:
Community is fixing spiffy binary to work properly with larger files, although build process already has spiffs_rom.bin build process.

anakod, 2015-05-11:
May be short example for start?

kireevco, 2015-05-25:
Currently files can be added by creating a 'files' directory, and placing files in it.
Also make sure spiffy binary is also available in the path.

closed 2015-05-25 by kireevco

MrZANE42, 2016-04-03:
Hi anakod/Sming,

I'd like to add you to my professional network on LinkedIn.

- Jimmy

Confirm that you know Jimmy: https://www.linkedin.com/e/v2?e=0-imkm2c97-z5&t=ssuw&tracking=eml-guest-invite-cta&ek=invite_guest&sharedKey=3WDSIfYF&invitationId=6122385882734489601

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=0-imkm2c97-z5&t=lun&midToken=AQFmrFVhGEd36Q&ek=invite_guest&loid=AQFv0bAsKE-cJgAAAVPcVjblnbMSBmjllQlD4yuoiPOdj6itsze5UFmt2D80kH8lELlt1o1kAd07W9egJLbJ5rXXQMqOeJneLdxyBvHr67Uz8_I53Xdqy1yYd7Oi5ESR5W21S4qc6wmXHeal3sNlphn_7kRY6f6WExRqC3-2FvXBkvqq_AsIQl7f4MCtSFaEk8IG&eid=0-imkm2c97-z5

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=0-imkm2c97-z5&a=customerServiceUrl&ek=invite_guest

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

MrZANE42, 2016-04-03:
Join Jimmy Pedersen and over 300 million other professionals already on LinkedIn

Sign up to become Jimmy's newest connection.

Confirm that you know Jimmy: https://www.linkedin.com/e/v2?e=0-imkm2cu7-b&t=ssuw&tracking=eml-guest-invite-cta&ek=invite_guest&sharedKey=Qin3Ojj0&invitationId=6122385854909472768

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=0-imkm2cu7-b&t=lun&midToken=AQFmrFVhGEd36Q&ek=invite_guest&loid=AQFuHW8ooHBBbwAAAVPcVkadPp9XqqK1h0QnObg_qRk6SW_IClQF92LZh2R9QNDTrwkNvxLEP78SiHaIzD5qDzXGByBhQ30jTmBJOYBcQwew2GqGpHniHxDGsD4kKVwlmE0nEEQqoOL_0Yh3r5spbeplnTFYu2FV-BFlRlHK0afPAY8_faom9rvcz6azW4vVJ3la&eid=0-imkm2cu7-b

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=0-imkm2cu7-b&a=customerServiceUrl&ek=invite_guest

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

MrZANE42, 2016-04-07:
Jimmy Pedersen would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/e/v2?e=-e3q58r-imq0n0x9-hu&a=preRegInvite&ek=first_guest_reminder_01&li=14&m=hero&ts=accept_text&sharedKey=Qin3Ojj0&invitationID=6122385854909472768

View Jimmy Pedersen&#39;s profile: https://www.linkedin.com/e/v2?e=-e3q58r-imq0n0x9-hu&a=preRegInvite&ek=first_guest_reminder_01&li=3&m=hero&ts=profile_text&sharedKey=Qin3Ojj0&invitationID=6122385854909472768

Jag skulle vilja lägga till dig i mitt yrkesnätverk på LinkedIn.

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=-e3q58r-imq0n0x9-hu&t=lun&midToken=AQFmrFVhGEd36Q&ek=first_guest_reminder_01&li=16&m=unsub&ts=HTML&eid=-e3q58r-imq0n0x9-hu&loid=AQF2zV-V4l4pEgAAAVPwz2dJppMl_6CXacBrucqSx1q56LgKfW44RG5HqcUAxPOUWVB6BskKQvk5wc0QwX0eDgBRmkSiVNAbsIykfWzJl5J77yIJMWDSO85aUhbIKJHSSlLNbXXW9_mtANaMm5S34_92BFo5eFOiVQdAOE_DRwHrKkhKpKOV0euU3MZs6fOqHC94

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=-e3q58r-imq0n0x9-hu&a=customerServiceUrl&ek=first_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

MrZANE42, 2016-04-07:
Jimmy Pedersen would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/e/v2?e=-71yzvu-imq0oy3v-ha&a=preRegInvite&ek=first_guest_reminder_01&li=14&m=hero&ts=accept_text&sharedKey=3WDSIfYF&invitationID=6122385882734489601

View Jimmy Pedersen&#39;s profile: https://www.linkedin.com/e/v2?e=-71yzvu-imq0oy3v-ha&a=preRegInvite&ek=first_guest_reminder_01&li=3&m=hero&ts=profile_text&sharedKey=3WDSIfYF&invitationID=6122385882734489601

Jag skulle vilja lägga till dig i mitt yrkesnätverk på LinkedIn.

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=-71yzvu-imq0oy3v-ha&t=lun&midToken=AQFmrFVhGEd36Q&ek=first_guest_reminder_01&li=16&m=unsub&ts=HTML&eid=-71yzvu-imq0oy3v-ha&loid=AQEkqInKZa3u1gAAAVPw0HgrzbsnKWonp3qDsU--dtqAF2_wL2heowG22go-WqwruAmqEK-W-R4tty-D1P4ewS34ADYUavLYIjEsfCXdrj96PbNOTbxC5Tn6zdIw3UBJ_rOJnl4MXhOVA9R_H0NjgH7K-GdDkRro0RBNG_SxZRF8h0LKlyvELtgZ9P15IRkTagdy

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=-71yzvu-imq0oy3v-ha&a=customerServiceUrl&ek=first_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

MrZANE42, 2016-04-14:
Jimmy Pedersen would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/e/v2?e=-71yzvu-in09eqof-88&a=preRegInvite&ek=second_guest_reminder_01&li=13&m=hero&ts=accept_text&sharedKey=3WDSIfYF&invitationID=6122385882734489601

View Jimmy Pedersen&#39;s profile: https://www.linkedin.com/e/v2?e=-71yzvu-in09eqof-88&a=preRegInvite&ek=second_guest_reminder_01&li=3&m=hero&ts=profile_text&sharedKey=3WDSIfYF&invitationID=6122385882734489601

Jag skulle vilja lägga till dig i mitt yrkesnätverk på LinkedIn.

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=-71yzvu-in09eqof-88&t=lun&midToken=AQFmrFVhGEd36Q&ek=second_guest_reminder_01&li=15&m=unsub&ts=HTML&eid=-71yzvu-in09eqof-88&loid=AQEAclFO_hGydwAAAVQVSVqya87Q6sCvaKWcQyupgEp1VHfUrYgZNxNKuJHOgGE4whbjSQ4dQy-EwrhJ4sXSg6EnFfi7FNWs5bpw5FULl8obur94sE2hTbBjVlZjxCY_KHewSOaO2LYEBiFDkyCc7tPPdSbZTBrdb-0NDTx8o0RGbiFRwEqPlSre8avu69X1pZz8

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=-71yzvu-in09eqof-88&a=customerServiceUrl&ek=second_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

MrZANE42, 2016-04-14:
Jimmy Pedersen would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/e/v2?e=-e3q58r-in09eifc-xd&a=preRegInvite&ek=second_guest_reminder_01&li=13&m=hero&ts=accept_text&sharedKey=Qin3Ojj0&invitationID=6122385854909472768

View Jimmy Pedersen&#39;s profile: https://www.linkedin.com/e/v2?e=-e3q58r-in09eifc-xd&a=preRegInvite&ek=second_guest_reminder_01&li=3&m=hero&ts=profile_text&sharedKey=Qin3Ojj0&invitationID=6122385854909472768

Jag skulle vilja lägga till dig i mitt yrkesnätverk på LinkedIn.

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=-e3q58r-in09eifc-xd&t=lun&midToken=AQFmrFVhGEd36Q&ek=second_guest_reminder_01&li=15&m=unsub&ts=HTML&eid=-e3q58r-in09eifc-xd&loid=AQHJhAFezoiEhwAAAVQVSs4pBexoYbdYFKFrme7qd5KdF7jo7U3XGqNlz1N15qNDYjpg8q2yjNQBfI_hDS-lEY-TmGUCS_q58RAoHe5C1a5W-FaOT3apAxIgaNWzPO9m-4u39TqYI9x7IPlDCPb7KUyUA7-A9PmpCKfNCjaUVYlIkbj7OFzImMW-ZZlMe0heSoFJ

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=-e3q58r-in09eifc-xd&a=customerServiceUrl&ek=second_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.