SmingHub/Sming #28: OTA firmware flashing

kireevco, 2015-04-15:
I would like to add this feature request to be able to update firmware from an http source.

prussiap, 2015-04-15:
Possibly pushing the usually booting pin low (GPIO0) to accept firmware? or a way to trigger that via web interface.

mikewen, 2015-04-20:
It would be amazing to have OTA firmware flashing. 
Are you going to write a bootloader?

kireevco, 2015-04-20:
Currently we are collecting information on how is it done in other ESP firmwares and estimating overall complexity.
Feel free to post here your ideas / links to documentation or join us in [image]

kireevco, 2015-04-27:
@anakod I know you have good news on this :)

anakod, 2015-04-27:
Yes, I have :)
**OTA firmware flashing was completed.**
https://github.com/anakod/Sming/blob/master/Basic_AirUpdate/app/application.cpp

Please, test!
**NOTICE**: you should update to Espressif SDK v1.0.1 b1 before, previous version contains a IROM access bug.

MrZANE42, 2015-04-29:
Nice work Anakod!
The download/flash update works like a charm, tried it multiple times. But for some reason the cpu wouldn't always start to run the new code after (vector or wd)reset. I also tried removing the jump to the reset vector and just use the wd reset, no change. A hardware reset or power cycling always started the new program without problem. It could possibly be hardware related (I have an Olimex board (https://www.olimex.com/Products/IoT/ESP8266-EVB/open-source-hardware) with bigger ext. flash),but I haven't seen similar problem before. Any Ideas?

anakod, 2015-05-01:
I think it hardware related stuff. Try to reboot module *before* OTA updating and it will reboot without any problems. So I think what after firmwaring it start user program but still work in firmware flashing move, as result reboot doesn't help. After normal restart OTA and rebooting will work.

closed 2015-05-12 by anakod