SmingHub/Sming #30: Sming examples, sming core has to be cross platform

kireevco, 2015-04-16:
Sming examples, sming core has to be cross platform - able to build seamless on Windows, MacOS and Linux using gnu / mingw tools.

caquino, 2015-05-02:
Appears that some pieces of the Makefiles still platform dependent.

`c:/Espressif/utils/gen_appbin.exe out/build/app.out 0 0 0 0
make: c:/Espressif/utils/gen_appbin.exe: No such file or directory
make: *** [out/build/app.out] Error 1`

Also I think it's good or to add "if's" for environment defaults or to add to the documentation that the following environment variables needs to be exported for a successful build.

- XTENSA_TOOLS_ROOT
- SDK_BASE

kireevco, 2015-05-02:
Please make sure to switch to a right branch when testing. Also don't forget to set SMING_HOME and ESP_HOME environment variables. Make sure to refer to correspondent wiki section on analog/Sming.

closed 2015-05-03 by kireevco