SmingHub/Sming #43: Ftp upload crashes after first file

lawrence-jeff, 2015-04-25:
Seeing a few odd things with the FTP code. Using the vanilla example you can upload one file successfully (small html). When you attempt to upload a second small file you get a 150 Connecting response and the next command will give you two consecutive 200 OK responses. I can't get it to upload more than 1 (sometimes 2 files before it starts doing this)

Steps to reproduce on an ESP-01 with 512k flash module
Pull the latest FTPServer example and supporting files
Update SSID and password
Compile 
I do a flashinit and then a flash (I believe this Is necessary to clear any old fs remnants)
ESP comes up correctly
Using windows 8.1 built in FTP
Open IPAddress
authenticate with me, 123
lcd to a directory with some static HTML files
set binary mode
put file1.ext - it works you get a connected followed by a Transfer Started followed by a  transfer speed summary
put file2.txt - You immediately get a Transfer completed followed by a 200 OK
FTP is hosed at this point and you can't even do a dir, if you exit out and back in you will find the second file is a zero length.

Cut and paste from terminal window
ftp> open 192.168.0.20
Connected to 192.168.0.20.
220 Welcome to Sming FTP
User (192.168.0.20:(none)): me
331 OK
Password:
230 OK
ftp> lcd c:\temp\tinyweb
Local directory now C:\temp\tinyweb.
ftp> binary
250 OK
ftp> put index.html
200 OK
150 Connecting
250 Transfer started
ftp: 2866 bytes sent in 0.08Seconds 36.74Kbytes/sec.
ftp> put api.html
226 Transfer completed
200 OK
ftp> put api.html
150 Connecting
ftp> put api.html
200 OK
200 OK

alonewolfx2, 2015-04-30:
did you tried file upload with ftp client program like filezilla. i can upload many files (small or 30-40kb)

lawrence-jeff, 2015-05-01:
I will retest with filezilla and try to isolate any differences if it works

anakod, 2015-05-01:
Also you can check TotalCommander (should work), maybe it can alow to take additional information.

anakod, 2015-05-15:
Any news here?

alonewolfx2, 2015-05-15:
yes. its fixed with spiff init.

closed 2015-05-15 by anakod