SmingHub/Sming #49: Add checks for conflicting environment

kireevco, 2015-05-02:
Add checks for conflicting environment (cygwin, for instance)

anakod, 2015-05-02:
About what this issue? Can you discover, please.

kireevco, 2015-05-02:
Let's say if we are installing sming on a windows box that has cygwin or delphi and path contains wrong, incopatible toolset (make.exe, for example) - it won't let us to build anything properly (different version, older fork, etc.). I would like to check, if our environment is sane.

j0hncc, 2015-05-15:
FYI I am finding that sming(mingw) AND my existing CYGWIN projects/environment does all work properly including eclipse once I settled on system path in this order:  c:\mingw\bin; c:\mingw\msys\1.0\bin ; c:\cygwin64

kireevco, 2015-05-15:
thanks @j0hncc

kireevco, 2015-06-28:
Closing for now. I'd rather move on to docker that will solve these issues by default
(https://github.com/kireevco/sming-docker)

closed 2015-06-28 by kireevco