SmingHub/Sming #56: Timer very unprecise and unpredictable

HaknCo, 2015-05-10:
Hi,
I tested the Basic_Blink example and compared the output against the Blink example from the Arduino IDE port (version from 7.May). Both examples toggle PIN2 every 500 ms. You can see output below.

The Arduin Blink looks like this. Signal looks as expected.
[image]

Now check this out. Totally weird.
[image]

 I also tried the pull request from hreintke, who updated Timer to use delegates. 
Same result.

Any suggestions ?

Regards

anakod, 2015-05-10:
I'm sure this isn't timer issue, something else affects on this pin in same time. You work this original example, yes? At first, please try to change selected pin number.

tonycorona, 2015-05-11:
Maybe I'm having the same issue when compiling the Sming blink example, the led blinks but very unregular, normal blink example works okay. I think it is because the ESP is constantly rebooting. I can compile and flash al the normal examples using eclipse and they work okay, i can also compile and flash all the Sming examples without error but they don't work? Only the blink example works sort of... ;) I also noticed the blue led on the ESP constantly blinking?

tonycorona, 2015-05-11:
Upgraded SDK to v1.0.1b1 and now it's working okay :+1:

anakod, 2015-05-11:
Yes, it can be related to the https://github.com/kadamski/esp-lwip/issues/7
Unfortunately, author doesn't solve last SDK compatiblity issues yet.

anakod, 2015-05-15:
Can be closed?

HaknCo, 2015-05-21:
Just checked out last version. Problem disappeared. Blink example works as expected.
Thanks.

closed 2015-05-21 by HaknCo