SmingHub/Sming #59: Help

wejedge, 2015-05-12:
Hi,
is there a way to use a callback after the response has been sent?
i.e. so that the list can be deleted?

Thanks.

Vector<String> list;

void onSendList(HttpRequest &request, HttpResponse &response)
{
     debugf("onSendList\r\n");

     JsonObjectStream* stream = new JsonObjectStream();
     JsonObject& json = stream->getRoot();

     json["count"] = list.count() / 2;

     for (int i = 0; i < list.count(); i += 2)
     {
        json[list[i + 0].c_str()] = list[i + 1].c_str();
     }
 
     response.sendJsonObject(stream);  
}

anakod, 2015-05-12:
1. You should save list in global variable
2. JSON requrie using static string variables (json lib doesn't copy pointers, just reference to them)
3. Very soon I will provide WiFi Network selection example

wejedge, 2015-05-12:
> You should save list in global variable
> JSON requrie using static string variables (json lib doesn't copy pointers, just reference to them)
> Very soon I will provide WiFi Network selection example

Hi,
thanks for ultra fast response.
Yes I thought about using a global, but I’m unsure how big the list is it also change's over time and don’t want to waste memory.
Maybe any other ideas?

Thanks

anakod, 2015-05-12:
I don't know you task, so I can't provide solution. Also I'm sure what without saving (to memory/to file/etc..) yo can't use data anyway.

wejedge, 2015-05-12:
I’m collecting data from a sensor (adc).
It’s not a regular period, sometimes more samples than others. The samples are recorded in ram and frequently sent to the browser.

anakod, 2015-05-13:
Main problem is what JSON doesn't save copy of your data. If you store integers/doubles/bools/etc.. (as keys or as a values, no difference) it will be saved in JSON library memory, but if you save strings, it just store pointer to you external object. So if you data can be represented as number - just save numbers. If not - additional work will be requried.

wejedge, 2015-05-14:
Ok understood thanks for the advice.

kireevco, 2015-07-27:
Closing for now

closed 2015-07-27 by kireevco