SmingHub/Sming #60: DateTime library

danielnilsson9, 2015-05-13:
Any chance having the comple arduino time library included so we could have the ability to keep track of time, it uses millis() to update time, and then we can sync agains maybe NTP every now and then.

The DateTime library that is included does not help us keep track of time.

https://github.com/PaulStoffregen/Time

anakod, 2015-05-13:
1. Please look this example of real time calculation: https://github.com/anakod/Sming/tree/master/MeteoControl
2. Can you discover more details about how you want to use it/how it should to work?

In my opinion, DateTime - is a *data type* (like numbers, strings and etc..), while actual time calculation is a task/method/.. It can be combined, of course, but I want to have some high level abstractions here also.

danielnilsson9, 2015-05-13:
Yep you are right, its probably a good idea to keep DateTime as just a data type. 

Anyway since I realized it was only a few lines of code I wrote a class SystemClock to keep track of time with NTP sync.

Thank you for this project, I guess we can close this issue again.

closed 2015-05-13 by danielnilsson9

anakod, 2015-05-13:
If it will be useful you can make pull-request to main branch to share you expirience with all other users :)
I think what it can be derived class (with using RTC clock), for example.

anakod, 2015-05-13:
Also if you can share your NTP code it will be great too, I think.

danielnilsson9, 2015-05-14:
Yes, using the RTC might be a good idea, but I'm not sure its very usefull. It seems to be just a counter like get_system_time() with the only difference that it keeps running during deep sleep, which may be good but I have read people not getting that to work.

There is however a 4 byte rtc memory which I guess could be used to store time in seconds (unix time) upon syncronization and then having time represented by whats in that memory + calculated time from system_get_rtc_time(). Care must be taken how to handle when RTC overflows though, its a uint32 counter. Not sure how to handle that if it happens during deep sleep for example, e.g. how should we no when we come back how many times the RTC has overflown during sleep?

Easiest solution would be to just do a new NTP sync on boot then I suppose, but then the RTC does not give us much benefit over using the regular get_system_time(). Also all ESP my not be conencted to internet and none of those solutions would work for these devices.

I will see if I can make a pull request (I'm new to github) with NtpClient and SystemClock.

It just a SNTP client though, since great accuracy wasn't needed in my case.

anakod, 2015-05-14:
NtpClient will be great example, thank you!