SmingHub/Sming #61: Error Function 'os_printf' could not be resolved, Symbol 'AUTH_OPEN' could not be resolved

tinamore, 2015-05-14:
Hi
I build project Basic_WiFi, and show error:
- Error Function 'os_printf' could not be resolved
- Invalid arguments WifiAccessPoint.config bool config(String, String, ?, bool, int, int)
- Symbol 'AUTH_OPEN' could not be resolved
Please see shotscreen:
http://s26.postimg.org/4trava9e1/55547fa7bf3e5.png

Please tell me, How to fix the above errors
Thanks

anakod, 2015-05-14:
Error list is wrong, please look **Console** window at the bootom.

tinamore, 2015-05-14:
yes, i see,
But how to fix errors
It took me 5 hours to full installation but still not working..

anakod, 2015-05-14:
How you install? Chocolatey or manual installation? Which OS?
If console output doesn't contains any errors, than this isn't real error just code checking & autocomplete detection mistake.

tinamore, 2015-05-14:
First, thank you for your support
I manual installation on Windows 7 64bit, There is only a difference in step 4.
I can not foud folder mingw64, I only see MinGW in C:/MinGW, This may be the cause of errors.
How do I install the mingw64 ?

anakod, 2015-05-14:
1) You doesn't use Chocolatey?
2) MinGW was already installed before?
3) You should set actual paths in variables, if MinGW installed in different folder - use that.

anakod, 2015-05-14:
Try to add paths:
* C:\MinGW\bin
* C:\MinGW\msys\1.0\bin

tinamore, 2015-05-14:
Yes,
C:\MinGW\bin -> exits
C:\MinGW\msys\1.0\bin -> exits
and i try add to Paths, But the error is still show.
----
In step 4, i Run mingw-get-setup.exe. But I don't know how "Add install support for the graphical user interface".
There is article about "Add install support for the graphical user interface" ?

tinamore, 2015-05-14:
I have just fix the error
The problem is the wrong path Espressif SDk,

The default path Espressif SDk in the Sming is  "C:\Espressif\sdk", While the default path when installing Espressif-ESP8266-DevKit-v2.0.4-x86.exe is "C:\Espressif\ESP8266_RTOS_SDK". I just changed the name "C:\Espressif\ESP8266_RTOS_SDK" to "C:\Espressif\sdk" =>The problem is resolved

Thanks you suppport.
I have a question, how to show Serial Console in Elipse LUNA?

anakod, 2015-05-14:
I understand your problem. Better rename **ESP8266_SDK**, not ESP8266_RTOS_SDK to "sdk"
I'm using separated serial application - Termite

tinamore, 2015-05-15:
Do you have any tricks to optimization performance for ESP8266 (I am using ESP8266-01)?

I try http server with: Nodemcu, ESP8266 Arduino, and Sming. But all are very slow web response.

For example the code toggles the led via http server. Click the button on, off LED in the web http://ipESP8266/, sometimes need a second, sometimes take several seconds, sometimes no response (Status browser: Connecting ...)

I have tried HttpServer_AJAX of Sming, I upload files from web/dev to FTP server of esp8266. However when running http://ipOfEsp8266/, it ran very slow, Not even load the js, css.

I think it will respond quickly, because the code runs as simple.

anakod, 2015-05-15:
No, it's wrong. You should upload web/build folder. Also, please, check files after uploading.

danielnilsson9, 2015-05-15:
Might be related to your router (if you are using station mode). Had the exact same problem before with very unstable connection between rounter and ESP. Ping times were consistently over 300ms and no http server did work any good.

I got a ASUS RT-N56U which was running custom firmware. I den reflashed with the default firmware and it started working, ping times under 2ms now.

But no other devices had problems with the router before so not sure why it was happening.

anakod, 2015-05-15:
>  I den reflashed with the default firmware and it started working, ping times under 2ms now.
Reflashed router, yes?

danielnilsson9, 2015-05-15:
Yes exactly!

anakod, 2015-05-15:
This issue can be closed?

danielnilsson9, 2015-05-15:
I'm not the OP, you sould propably wait for @tinamore replay  he may not have exactly the same problem as I have had before. If you were talking to me?

tinamore, 2015-05-15:
Thank you for support. Sming is great project.

kireevco, 2015-05-15:
@anakod thank you for updating Wiki https://github.com/anakod/Sming/wiki/Windows---Manual-Installation

@tinamore I will close this ticket for now. Feel free to reopen it if you have more questions.

closed 2015-05-15 by kireevco