SmingHub/Sming #67: Timer issue, large interval

danielnilsson9, 2015-05-15:
Hi

Think i found an issue with the timers. Setting a long interval causes issues, for example 10 minutes, 10*60*1000 = 600000ms. Its not related to overflow, but that would happen if setting a longer interval than 71 minutes in the way the Timer class is implemented, intervalUs = intervalMs * 1000 should cause overflow in the uint32 if intervalMs > approx 71 minutes.

Anyway thats not the issue here:
If setting timer interval repeating 10 minutes the callback will be called twice in a row every approx 340 seconds, which is strange. Even if doing it manually with the ets_timer functions the same issue persist in Sming.

BUT: if not using Sming framework (only using ESP SDK) and doing it with ets_timer functions there is no issue at all, callback called every 600 seconds as expected. (Same SDK version used)

Have been looking but cannot find the cause, using the exact same ets_timer code without using Sming works, but when using Sming it doesn't.

Any ideas? Seems like a tricky one and very boring to test... :(

anakod, 2015-05-15:
Is it on last version (yersterday\today changes)?

danielnilsson9, 2015-05-15:
Updated aprox 12 hours ago, haven't updated since then. It was when you merged my pull request about bug fixes in NtpClient. I will try.

anakod, 2015-05-15:
I have commited new changes now. It will be uint64_t to cover all cases.

anakod, 2015-05-15:
Miss one moment, fixed.

danielnilsson9, 2015-05-15:
Yes thats a good idea, will fix that 71 minute problem. But I don't think it will solve the other one which happens even if not using the Timer class.

If you run the SystemClock_NTP example and set update interval to 10 minutes you should set that it will request update every ~5 minutes with two callback each time   If its not something wrong with my board..

anakod, 2015-05-15:
I think I know difference - Sming using system_timer_reinit() at the start for PWM.

j0hncc, 2015-05-15:
I did some pretty thorough testing of different values.
I can't seem to find my notes!  
But my conclusion was very clear, there is a modulo going on something 450000-500000 msec. (Can't remember exact number though I did determine it).  
Eg  460000 would delay correct 7.6 min but the 465000 would delay only 5 seconds!
Eg 925000 would also delay 5 seconds.

danielnilsson9, 2015-05-15:
Yes, must be the system_timer_reinit(), putting that in my test which wasn't using Sming causes the same problem. Then it must be a bug in the ESP SDK then?

kireevco, 2015-07-27:
PR ws Merged. @danielnilsson9 please confirm it is working

danielnilsson9, 2015-08-07:
@kireevco @anakod Nope i don't believe it has been fixed. Espressif said that this was due to system_timer_reinit() as mentioned above but they didn't seem to have any intention to increase the maximum interval. But I believe it should definitely be possible for them since the os_timer is a software timer as I have understood.

This is the reply I got:
>Default timer is millisecond timer and the maximum value is 6871947ms.
>
>After call system_timer_reinit,the timer will change to microsecond timer and the maximum value of >os_timer_arm() is 429496ms.

>Any more questions, please let us know.

Which I haven't found documentet anywhere so that is good to know.

Anyway, best thing would probably be to add a workaround to Sming Timer class so it can handle larger intervals which would be a fairly simple fix, just split the the interval if it is too large and have a more frequent callback (within allowed interval) and a counter.

Any thoughts on that? I can do it and make a pull request. Also the ntp client demo should be cleaned up again as it is very confusing right know, showing all possible ways it can be used makes it harder to understand.

Also there is more bugs in NtpClient now, DEFAULT_INTERVAL is specified in milliseconds, constructor taking interval in seconds but DEFAULT_INTERVAL is passed for first and seconds constructor which must be wrong. 

Also server ip is saved for an infiniate amount of time after succesfull dns lookup in NtpClient which may not be a god idea (server ip could possibly change). I suppose the DNS implementation caches the lookup for a period of time of which after it makes a new lookup, we should make use of that, its a quick fix.

Any thought on these things, should I prepare a PR?

hreintke, 2015-08-09:
@danielnilsson9 : I made the update to ntpclient to make it generic.
Agree on the bug in NtpClient., maybe the best is to change
#define NTP_DEFAULT_AUTO_UPDATE_INTERVAL 600000 // 10 minutes
to
#definan e NTP_DEFAULT_AUTO_UPDATE_INTERVAL 600
The all used numbers are in seconds and inetrnally recalculated to ms for calling timer.
Do you want to make the PR or should I do it.
For Sever <> IP DNS lookup. I think best is to include an connect(String server, int port) function in udpconnection and the make use of that in ntpclient.
The we make it identical to tcpclient and usable for all udpclient usage.
Same question : do you want me to create a PR of will you do it

danielnilsson9, 2015-08-09:
Hi yes, that sound good, already made the changes so I will post a PR today or tomorrow for a few things we can review then. If the udpconnection class already does dns lookup then that code can be removed alltogether.

I have also added a timeout so a new request will be made  if a response isn't received in 30 seconds (maybe we should have binary exp backoff instead?)...

danielnilsson9, 2015-08-09:
Not sure how to do in the UdpConnection class, if we have connect which takes a domain the connect is not blocking meaning we need to have a callback to tell the user when its ok to send, we we try to send directly after connect() the dns lookup may not have been done.

I will make a PR with the changes I have made to NtpClient and we can think about the rest later.

hreintke, 2015-10-03:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-03 by hreintke