SmingHub/Sming #70: PWM unstable/flickering

zigarrre, 2015-05-15:
When driving a LED with PWM it flickers. This is probably caused by timing issues in the PWM implementation which cause a varying duty cycle. The lower the duty cycle is set the more drastic will be the flickering.

Minimum sample to reproduce the problem:
```c++
#include <user_config.h>
#include <SmingCore/SmingCore.h>

void init()
{
        analogWrite(2, 50);
}
```
LED connected between GPIO2 and GND.

tprochazka, 2015-06-08:
Here is interesting blog post about PWM
http://tech.scargill.net/esp8266-pwm-on-one-output/

tprochazka, 2015-06-13:
Has ESP real HW PWM on some pin or everything is only emulated?

I tried it and flickering is minor problem, much bigger problem is that ESP is completely unresponsive during PWM.

zigarrre, 2015-06-13:
The ESP has a hardware PWM peripheral (on pins 12-15, which the link you posted is using) but the PWM implementation in Sming is software PWM. The problem is the suboptimal implementation in Sming.

The software PWM implementation in NodeMCU (which i'm using now) works perfectly fine without any flickering, even at the lowest duty cycle.

tprochazka, 2015-06-15:
According
https://github.com/esp8266/Arduino/issues/424#issuecomment-111876468
ESP has no HW support for PWM

tprochazka, 2015-06-15:
SDK 1.1.1 has some new driver for PWM
http://bbs.espressif.com/viewtopic.php?f=5&t=554

kireevco, 2015-06-28:
@zigarrre can you please see if 1.1.1 fixes that? Thank you!

hreintke, 2015-06-28:
@kireevco : I don't think sming uses the new PWM drivers yet. They are in libpwm.a and that is not used by sming

zigarrre, 2015-06-28:
@kireevco @hreintke : hreintke is right, Sming uses it's own software PWM implementation (see https://github.com/anakod/Sming/blob/master/Sming/SmingCore/PWM.cpp).

anakod, 2015-07-01:
Current PWM implementation needs update. Good example: https://github.com/sh4d0w12/esp8266_IRremote/pull/1

hrsavla, 2015-07-23:
@anakod Sming's PWM.cpp is using ostimer. Bellow is written in latest SDK document Page 15.
"Please be noted that os_timer APIs listed below are software timer, they are executed in task, so timer callback may not be precisely executed at the right time, it depends on priority. If you need a precise timer, please use hardware timer which will be executed in hardware interrupt, refer to hw_timer.c. "
Also latest SDK PWM api seems to be using hw_timer. (bellow is from page18)
"APIs in hw_timer.c can not be called when PWM APIs are using, because they are using the same hardware timer."
So we should make change Sming PWM to use hw_timer. I think this will solve "Flicker" problem.

kireevco, 2015-07-27:
@hrsavla do you think you could add that change vie PR?

crosofg, 2015-08-17:
Is flickering problem solved?

hrsavla, 2015-08-17:
Software PWM generated in Sming will generate flickers as timers are of not higher priority and some Network routines will interrupt it. Solution is to use ESP sdk pwm  api or write whole new PWM class using hardware timer. 
I am able to make Esp sdk PWM to work with sming. I still need to format the code in Sming way Class files. I will do it and make a PR within 6 hours. (I will let you know).

crosofg, 2015-08-18:
Okay.

crosofg, 2015-08-18:
Hi did you complete it ?

crosofg, 2015-09-06:
@hrsavla Did you remove your implementation of hw pwm?

hrsavla, 2015-09-06:
I am resubmitting it in an hour as per git flow.

crosofg, 2015-09-14:
How can I combine two features? I want to implement rboot and Hw pwm. what changes needs to be done?

hreintke, 2015-10-30:
In progress. See PR #375

closed 2015-10-30 by hreintke