SmingHub/Sming #72: Mqtt client is Not stable

tinamore, 2015-05-16:
Hi
I am flash project name "MqttClient_Hello", and before flash I've been editing
From
procTimer.initializeMs(20 * 1000, publishMessage).start(); // every 20 seconds
To
procTimer.initializeMs(2 * 1000, publishMessage).start(); // every 2 seconds

I noticed that the action publish run stable 1 minutes then stopped.
If I edit the publish every 5 seconds then run stable for 3 minutes then stopped, next  3 minutes then ESP8266 reboot.

I think there are problem in memory, as the video below:
https://www.youtube.com/watch?v=srWKOh7IR7U

Is there a way to run most stable MQTT ?
Thanks.

kireevco, 2015-06-28:
@tinamore what's going on with this, do you see any improvements with the latest sming version?

Thanks.

tinamore, 2015-06-29:
I have just test version Sming 1.2
Mqtt is still NOT stable.
1. When broker restart, mqtt NOT try reconnect. I try add code below, mqtt will try reconnect:
```
class ReconnctingMqttClient2: public MqttClient{
    using MqttClient::MqttClient; // Inherit Base's constructors.

    void onError(err_t err)  {
        close();
        connect("esp8266");
        subscribe("main/status/#");
        return;
    }

};
// MQTT client
// For quickly check you can use: http://www.hivemq.com/demos/websocket-client/ (Connection= test.mosquitto.org:8080)
ReconnctingMqttClient2 mqtt("mqttserver.com", 8000, onMessageReceived);
```
But Case 2 below: mqtt still error
2. When my internet disconnect and reconnect, mqtt NOT try reconnect.

I'm using mqtt example in SDK. MQTT client run very good. It always try to connect in all cases, run very stable.

I would like to use the Sming to simple for make project. I hope Sming mqtt will stable, because Sming is a easy code project for build esp8266

anakod, 2015-07-01:
Hello. Examples provided with Sming - just usage samples, not final, ready to use solutions. I have added auto-reconnect to MQTT example, please try it.

This functionality allow to restore working after internet, server or WiFi connection lost.

kireevco, 2015-07-01:
Thank you, @anakod

tinamore, 2015-07-01:
Thank @anakod
But still have error.
I see you fix reconnect when publishMessage(), but subscribe still will lost waiting receive message (I've just test).
Can you implement https://github.com/tuanpmt/esp_mqtt into the Sming project?
Because this tuan mqtt library is stable.
Mqtt is the important method of Iot, I think it need: stability, reliability, mqtt is must always active in all cases.

anakod, 2015-07-01:
> I see you fix reconnect when publishMessage(), but subscribe still will lost waiting receive message (I've just test).
What is difference? If publishing isn't needed (or provided too rarely), you can just make this conection check in separated timer with any requried interval.

kireevco, 2015-07-27:
@tinamore can you please provide an update for this?

kireevco, 2015-08-18:
Closing due to inactivity. @tinamore feel free to reopen.

closed 2015-08-18 by kireevco

sguilly, 2015-11-04:
An alternative exist for the MqttClient with the integration of the tuan mqtt lib.

https://github.com/SmingHub/Sming/pull/401