SmingHub/Sming #73: MEM CHECK FAIL!!!: After flashing most of the examples

kireevco, 2015-05-17:
I know a lot of people have this, let's investigate this please and bring serial output to order.
```
;ld��|�$�|�$�c|ǃ��{�c�b��og�l'o���#x�${l{$8�o��$b'�|��$�c��g'�lćd`�g�lo;���g��lx�o�;�����#g�|d�l�#��ng��d`�g$�d`gs�ۓn��``�g��#�gd���'o��lx�n�{�����d��cg�|쏇p�c��'o�d�$`��dgs�ۓo#l$`;��gc$l ��ՁMEM CHECK FAIL!!!
```

anakod, 2015-05-27:
May be try to clean flash with empty512kb.bin?

HappyCodingRobot, 2015-05-27:
Have tried it, but still shows the 'MEM CHECK FAIL!!!' message.
One module shows this, the others not.

kireevco, 2015-05-27:
What people on forums say?

kireevco, 2015-07-03:
I also heard that this could be caused by screen client only. Can anyone test other terminal clients please?

kireevco, 2015-07-06:
I with recent sdk and sming I don't seem to see this issue

closed 2015-07-06 by kireevco

alon24, 2015-07-06:
I got this today, so i wrote my esp as dead, but u say that recent sdk is ok?

Which sdk? the one in docker?
(it happened on my 12e)