SmingHub/Sming #75: How to keep LED connected to between GPIO2 and GND while reboot

tinamore, 2015-05-18:
I have a basic problem.

I have flash project Blinky, After the flash, the LED blinks normally (LED connected to between GPIO2 and GND) 

But When my ESP01 is restarted (I've removed GPIO0 from GND to switch to normal mode) then the problems: LED Not blink and always bright.

If I temporarily disconnect the LED circuit, the board will boot normally, and after reconnecting, my program works as desired (LED blinks normally).

Is there anything I can do to keep LED connected to between GPIO2 and GND while reboot ? (because of the fact the LED always connected between GND and GPIO2)

anakod, 2015-05-19:
Hello. It isn't related to Sming, and was a part of ESP boot modes. You can revert LED connection to 3V3 instead of ground, and keep level on GPIO2 high during boot, for example. You can find more details on ESP hardware discussions.

tinamore, 2015-05-19:
I have understood. Thank you very much.

closed 2015-05-25 by tinamore