SmingHub/Sming #77: Api documentation required

kireevco, 2015-05-19:
We need to have a clear documentation on sming methods, to see what's available and how it is supposed to work.

marvinroger, 2015-05-19:
#76 would come along with this.

AlmightyFrog, 2015-05-19:
Might sound simple stupid, but what about good old doxygen compatible comments per function or such?
That way it is easy to maintain in code directly but you still can generate some quite good documentation to read and search.

kireevco, 2015-05-19:
I think it is a very good idea.

AlmightyFrog, 2015-05-23:
I guess we should find a common agreement on which format to use.
Here you can see some example I tied out: https://github.com/AlmightyFrog/Sming/commit/9c65f5f36e6e38bf5232b7334cefd9e6d665a339
Just install doxygen if not yet, go into Sming/ and enter "doxygen" to start generation. Afterwards you can open HTML from Sming/doc/

Not really happy with result, but maybe it helps some one else to come up with better suggestion.

One part where I'm really not sure is whether it's in a project like this better to comment in header or in source file.

kireevco, 2015-06-28:
@AlmightyFrog Higly recommend to try NetBeans, it support Doxygen out of the box (as well as really nice code assistance)
[image]
[image]

kireevco, 2015-07-27:
@AlmightyFrog I like your format. In fact this is a pure doxygen format. 
If @anakod has no objections, let's continue in this direction.

I would like to ask for all core commits to contain doxygen documentation.

closed 2015-07-27 by kireevco

riban-bw, 2016-01-03:
Sorry to comment on a closed issue but I don't think this is fully resolved. The API is not fully documented and running doxygen against the Sming source does not give a particularly useful output. It would be advantageous to provide well documented API. (I have spent the past couple of weeks trying to understand Sming - it shouldn't be this difficult!) I would be happy to add some doxygen compatible docs to the source code (headers) but this needs to be coordinated so that we reduce duplicated or conflicting effort and use consistent documentation convention. Who is best placed to coordinate such a task (or empower someone to do so)?

It may also be advantageous to build the API documentation and host it online (within the wiki?) which may promote the project and facilitate developers.

Cheers Walt

kireevco, 2016-01-03:
@riban-bw Someone needs to start doing this, and I feel like it should be you, @riban-bw. Since no one else was actually doing a thing in that direction - they will follow you, be a leader. Please, don't spend time on coordination that much, as you see in this ticket, it doesn't really help to make things happen. Lead by example, that's very simple and agile.

What you can do - go to gitter and tell everyone you are documenting a specific class/classes. That's it.