SmingHub/Sming #78: NodeMCU-v2 ESP8266 development kit

tprochazka, 2015-05-20:
Does somebody already trying use this board?
http://www.seeedstudio.com/depot/NodeMCU-v2-Lua-based-ESP8266-development-kit-p-2415.html

Flashing works correctly but, program doesn't start, maybe because of not enoug power from USB, but I'm using USB 3 port and with original firmware was LED flashing.

Or it require some different setup?

kireevco, 2015-05-20:
Hi, not confirmed, but the problem seems to be related to the flash size of that board.

tprochazka, 2015-05-20:
According the
https://github.com/nodemcu/nodemcu-devkit-v1.0
ESP-12-E core it has 32Mbits (4MBytes) flash version.

What should be changed to test it?

tprochazka, 2015-05-20:
Similar topic
https://github.com/esp8266/Arduino/issues/292

tprochazka, 2015-05-23:
According the 2A-ESP8266__IOT_SDK_User_Manual__EN_v1.0.1.pdf documentation, flash memory with 1024kB and more kB has different addressation, you can see it here

http://bbs.espressif.com/viewtopic.php?f=10&t=175

I found that in the Sming makefile possibility to set SPI_SIZE=4096
But it doesn't help. esptool still use old addresses.
I not found way how esptool grab the right addresses.

I found that you change parametrs of gen_appbin.exe according the SPI_SIZE
So it should generate right bin for the desired SPI_SIZE
But where is deffined parametrs for esptool?

tprochazka, 2015-05-23:
I found it in Makefile-project.mk
There are all address hardcoded

tprochazka, 2015-05-23:
But change
```
flashinit:
	$(vecho) "Flash init data default and blank data."
	$(ESPTOOL) -p $(COM_PORT) write_flash 0x3fc000 $(SDK_BASE)/bin/esp_init_data_default.bin 0x3fe000 $(SDK_BASE)/bin/blank.bin
```
doesn't help

tprochazka, 2015-05-23:
So it is already solved on Arduino IDE implementation:
https://github.com/esp8266/Arduino/issues/292#issuecomment-104898668
But I'm not able to adopt this solution to Sming

SPI_MODE=DIO
SPI_SIZE=4096

doesn't work here :-(

anakod, 2015-05-24:
I'm need ESP with large flash I think. Now I have only 512 Kb modules.

tprochazka, 2015-05-24:
I have one from here
http://www.seeedstudio.com/depot/NodeMCU-v2-Lua-based-ESP8266-development-kit-p-2415.html
It's great module for dev.

It's possible to buy it also here
http://www.aliexpress.com/wholesale?catId=0&initiative_id=AS_20150523232940&SearchText=NodeMCU+v2

Or ESP-12E module itself
http://www.aliexpress.com/item/ESP8266-serial-WIFI-industry-milepost-type-ESP-12/32240399719.html

rosiste, 2015-06-03:
Hi guys, any news about how to generate right bin & flash ESP-12e on NodeMCU v2?

anakod, 2015-06-03:
I think we should update SPI Flash parameters in make file, but I'm still waiting this board. May be somebody can try?

https://github.com/anakod/Sming/blob/master/Sming/Makefile-project.mk#L317

anakod, 2015-06-04:
http://bbs.espressif.com/viewtopic.php?f=7&t=25 - error information. Can somebody change serial speed to bootloader default (74880 baud) and post here full log?

rosiste, 2015-06-05:
Build log with 
```
09:32:19 **** Build of configuration Default for project Ex DHT ****
mingw32-make.exe -f C:/Projects/ESP8266/Sming/Humidity_DHT22/Makefile rebuild 
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out
Running objcopy, please wait...
Objcopy done
Run gen_appbin
No boot needed.
Generate eagle.flash.bin and eagle.irom0text.bin successully in folder out/firmware.
eagle.flash.bin-------->0x00000
eagle.irom0text.bin---->0x9000
Done
```
And flash log on 74880 baud:

```
09:36:05 **** Build of configuration Default for project Ex DHT ****
mingw32-make.exe -f C:/Projects/ESP8266/Sming/Humidity_DHT22/Makefile flash 
/c/Projects/ESP8266/ESP.SDK/utils/esptool.exe -p COM5 -b 74880 write_flash 0x00000 out/firmware/eagle.flash.bin 0x9000 out/firmware/eagle.irom0text.bin 0x38000 out/firmware/spiff_rom.bin
Connecting...
Erasing flash...

Writing at 0x00000000... (4 %) 
Writing at 0x00000400... (8 %) 
Writing at 0x00000800... (12 %) 
Writing at 0x00000c00... (16 %) 
Writing at 0x00001000... (20 %) 
Writing at 0x00001400... (24 %) 
Writing at 0x00001800... (28 %) 
Writing at 0x00001c00... (32 %) 
Writing at 0x00002000... (36 %) 
Writing at 0x00002400... (40 %) 
Writing at 0x00002800... (44 %) 
Writing at 0x00002c00... (48 %) 
Writing at 0x00003000... (52 %) 
Writing at 0x00003400... (56 %) 
Writing at 0x00003800... (60 %) 
Writing at 0x00003c00... (64 %) 
Writing at 0x00004000... (68 %) 
Writing at 0x00004400... (72 %) 
Writing at 0x00004800... (76 %) 
Writing at 0x00004c00... (80 %) 
Writing at 0x00005000... (84 %) 
Writing at 0x00005400... (88 %) 
Writing at 0x00005800... (92 %) 
Writing at 0x00005c00... (96 %) 
Writing at 0x00006000... (100 %)
Erasing flash...

Writing at 0x00009000... (0 %) 
Writing at 0x00009400... (1 %) 
Writing at 0x00009800... (1 %) 
Writing at 0x00009c00... (2 %) 
Writing at 0x0000a000... (2 %) 
Writing at 0x0000a400... (3 %) 
Writing at 0x0000a800... (3 %) 
Writing at 0x0000ac00... (4 %) 
Writing at 0x0000b000... (5 %) 
Writing at 0x0000b400... (5 %) 
Writing at 0x0000b800... (6 %) 
Writing at 0x0000bc00... (6 %) 
Writing at 0x0000c000... (7 %) 
Writing at 0x0000c400... (7 %) 
Writing at 0x0000c800... (8 %) 
Writing at 0x0000cc00... (9 %) 
Writing at 0x0000d000... (9 %) 
Writing at 0x0000d400... (10 %) 
[... 157 lines cut ...]
Writing at 0x00034c00... (99 %) 
Writing at 0x00035000... (100 %)
Erasing flash...

Writing at 0x00038000... (6 %) 
Writing at 0x00038400... (12 %) 
Writing at 0x00038800... (18 %) 
Writing at 0x00038c00... (25 %) 
Writing at 0x00039000... (31 %) 
Writing at 0x00039400... (37 %) 
Writing at 0x00039800... (43 %) 
Writing at 0x00039c00... (50 %) 
Writing at 0x0003a000... (56 %) 
Writing at 0x0003a400... (62 %) 
Writing at 0x0003a800... (68 %) 
Writing at 0x0003ac00... (75 %) 
Writing at 0x0003b000... (81 %) 
Writing at 0x0003b400... (87 %) 
Writing at 0x0003b800... (93 %) 
Writing at 0x0003bc00... (100 %)

Leaving...

09:36:47 Build Finished (took 42s.605ms)
```

rosiste, 2015-06-05:
On 74880 baud terminal apears much better (DHT_22 example). 

```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 21664, room 16 
tail 0
chksum 0xef
load 0x33333333, len 858993459, room 8 
```

rosiste, 2015-06-05:
Same with SIZE=512
```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 21664, room 16 
tail 0
chksum 0xef
load 0x33333333, len 858993459, room 8 
```

koppi, 2015-06-11:
Same problem here, howto fix?

anakod, 2015-06-14:
Hello, sorry, I'm still waiting my ESP12E board from China. If somebody have news/solutions related to this problem - please, notify us!

jeremychild, 2015-06-15:
I have an ESP12E (Nodekit v2) and I have tried a few methods, including updating the esptool.exe but to no avail. Works in Arduino ESP8266 so ill see how they are doing it.

anakod, 2015-06-26:
Almost fixed: https://github.com/anakod/Sming/commit/48cc0a46529ead3b3f2c199df84dec0985e5b078
Ready to use, but need more deep check.

Problem was in an awful Makefile. Also I have completed some basic cleaning.

anakod, 2015-06-26:
I will be glad to hear any feedback about current build implementation on different ESP boards.

Squonk42, 2015-06-26:
Nope. Tried with the "Basic_Serial" example on (40MHz/QIO/4MB) NodeMCU-v2 ESP8266 development kit:

```...
out/build/app.out:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000554  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       000008b8  3ffe8560  3ffe8560  00000640  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00009a88  3ffe8e18  3ffe8e18  00000ef8  2**4
                  ALLOC
  3 .irom0.text   0002d4f6  40209000  40209000  00006a30  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         00005b31  40100000  40100000  00000ef8  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
/opt/esp-open-sdk/esptool/esptool.py elf2image out/build/app.out -ff 40m -fm qio -fs 32m -o out/firmware/
...
```

Then:

```23:35:27 **** Build of configuration Sming for project Basic_Serial ****
make flash 
pkill screen
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 230400 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x38000 out/firmware/spiff_rom.bin
Connecting...
Erasing flash...

Writing at 0x00000000... (3 %) 
...
Writing at 0x00006800... (100 %)
Erasing flash...

Writing at 0x00009000... (0 %) 
...
Writing at 0x00036400... (100 %)
Erasing flash...

Writing at 0x00038000... (6 %) 
...
Writing at 0x0003bc00... (100 %)

Leaving...

23:35:41 Build Finished (took 14s.35ms)
```

I have an up-to-date esp-open-sdk from https://github.com/pfalcon/esp-open-sdk.git

With "screen" @9600, 115200, 230400 bps, I still only have the startup garbage, no sign from app :(

tprochazka, 2015-06-28:
@anakod What I must change in compile configuration to make it work?
In you commit is a lot of changes, which SPI_MODE to you use for ESP-12E?

Squonk42, 2015-06-28:
It looks like the ESP-12E is using "QIO" ("Quad I/O"?) SPI mode and a 32MB (4Mbyte) Flash @ 40 MHz, here ar emy environment variables under Eclipse:

[image]

tprochazka, 2015-06-29:
According the ARDUINO IDE project ESP-12E need SPI_MODE=DIO, there are using this in the config file pro ESP-12E and it working, but not here, but if @anakod did some changes related to ESP-12E it should now work in some way.

mstempin, 2015-06-29:
OK, thanks, I had an ESP8266/Arduino that was outdated, found this:
https://github.com/esp8266/Arduino/issues/292#issuecomment-104884451

I will try asap.

Squonk42, 2015-06-29:
OK, thanks, I had an ESP8266/Arduino that was outdated, found this:
esp8266/Arduino#292 (comment)

I will try asap.

shansted, 2015-06-30:
Initially I couldn't get this to work with my 12E's. With DIO, 4MB set the software would start, and then crash as soon as it went to format the filesystem. I upgraded the esptool ( https://github.com/themadinventor/esptool ), and then all appeared OK, but the filesystem doesn't work. I cant delete files or upload any. fsformat also returned instantly, without actually formatting the filesystem.

If i flash the same code using the Flash Download Tool http://bbs.espressif.com/viewtopic.php?f=5&t=433), everything appears to work fine. Some issues with large files, but I've not tested this extensively yet.

This tool also lets you identify the flash and alter the transmit power of the wifi!!!

anakod, 2015-07-01:
Please, try again on last version, I have added WDT blocking

anakod, 2015-07-01:
### Please, be careful
ESP-12E use SPI_MODE=**DIO**

tprochazka, 2015-07-01:
So, you should change it in the build script because there is uppercase :-(
```
# SPI_MODE: QIO, QOUT, DIO, DOUT
SPI_MODE?=QIO
```
I tried use dio now with the latest sming from master and Blink example works!!!

But if I tried something more complicated like HttpServer_ConfigNetwork it print this to serial port

```
rl[00]lśž|[00]Ślŕ|[02][0C][0C][0C]Ś[0C]lě[0C]b|Ž‚[02]ě[12]’r’bŚ[0C]bŚňnnžlnnśâě[0C]b[1C]pŚŽlrlrlpňnŕ[10][02][0C][0C]‚[0C]l[0C][0C][0C][0C][0C][0C]b[0C]nâ|[02]ŚěěŽŽ[0C]bŚňnnî[00]lŚŽl`[02][12][12]nn[0C]l`[02][0E][02]nrŽ’’n[0C][0C]’[02]l`[02]pňnŕ[10][02][0C][0C]rŚśśâŕ[0C][0C][0C][0C]b[0C]nâ|[02]lělŽ[1C]p[0C]Śňnnî[00][0C][0C]l`[02][12][12]nn[0C]l`[02][0E][02]nrŽ’’n[0C][0C][12]lŽ[0E]lpňnŕ[10][02][0C][0C]rŚśśâŕlŽr€[0C]b[0C]nâ|[02][0C]ŚlŚŽbŚňnnî[00][0C][0C]l`[02][12][12]nn[0C]l`[02][0E][02]nrŽ’’n[0C][0C]r[0C]Ž[0E]l[0E]r’’n[0C][0C]r[0C]Ž[0E]lŚ‚[00]
MEM CHECK FAIL!!!
drlS+TCP connection
timeout updating: 70 -> 65535
Server timeout updating: 40 -> 90
+TCP connection
timeout updating: 70 -> 65535
Server timeout updating: 40 -> 900
fs.start: size:224 Kb, offset:0x44000

First init file system
sect_first: 44, sect_last: 7c

sect: 68
sect: 69
sect: 70
sect: 71
sect: 72
sect: 73
sect: 74
sect: 75
sect: 76
sect: 77
sect: 78
sect: 79
sect: 80
sect: 81
sect: 82
sect: 83
sect: 84
sect: 85
sect: 86
sect: 87
sect: 88
sect: 89
sect: 90
sect: 91
sect: 92
sect: 93
sect: 94
sect: 95
sect: 96
sect: 97
sect: 98
sect: 99
sect: 100
sect: 101
sect: 102
sect: 103
sect: 104
sect: 105
sect: 106
sect: 107
sect: 108
sect: 109
sect: 110
sect: 111
sect: 112
sect: 113
sect: 114
sect: 115
sect: 116
sect: 117
sect: 118
sect: 119
sect: 120
sect: 121
sect: 122
sect: 123
sect: 124
formated
mount res: 0

[W]sec 7d error
[W]sec 7d error
[W]sec 7d error
[W]sec 7d error
[W]sec 7d error
[W]sec 7d error
[W]sec 7d error
(newer ending loop with [W]sec 7d error)
```

Squonk42, 2015-07-01:
Same here!

It looks like the sector 7d is not formatted. I suspect that the WiFi stack is looking for some default values in the data partition, found it non formatted and crashes miserably.

I tried to perform an "initflash" to fix this, but the "$(SDK_BASE)/bin/clear_eep.bin" (EEPROM?) file is missing, and I cant find it anywhere else.

anakod, 2015-07-01:
You on linux? Please try "$(SDK_BASE)/bin/blank.bin" or "$(SMING_HOME)/compiler/data/blankfs.bin" instead.

alon24, 2015-07-01:
The fs.start=224kb, says you are NOT on 4M (12e comes with 4m - as much as I have it)

Try
SPI_SIZE ?= 4M
SPI_MODE ?= dio
SPI_SPEED ?= 80 
(SPI_SPEED is the missing part here)

anakod, 2015-07-01:
Yes, but if you only update SPI_MODE it still should work, just with old file system size.

alon24, 2015-07-01:
the only issue I have here is that it did not work for me on sdk111 112, until I did those 3 things, then it started to work and I am a happy camper now....

So I see that this bug is still open, so working or not, if not, try what I suggested?

anakod, 2015-07-01:
@alon24 you checked it on last version, yes?

alon24, 2015-07-01:
yes,  from.  testerday

Squonk42, 2015-07-01:
Using latest Sming just freshly pulled, updated esp_open_sdk with esp_iot_sdk_v1.1.2_15_06_12, tried both "$(SDK_BASE)/bin/blank.bin" and "$(SMING_HOME)/compiler/data/blankfs.bin", with following Environment variable set in global Preferences:
BOOT = none
COM_PORT = /dev/ttyUSB0
ESP_HOME = /opt/esp-open-sdk
SMING_HOME = ~/Sming/Sming
SPI_SIZE = 4M
SPI_MODE = dio
SPI_SPEED = 80
WIFI_PWD = my password
WIFI_SSID = my SSID

3basic_Blink" is working, but I am getting the same infinite "[W]sec 7d error" with "Basic_WiFi" example.

Any idea what I am doing wrong?

anakod, 2015-07-02:
I have update it again, please check.

anakod, 2015-07-02:
> So, you should change it in the build script because there is uppercase

Where?

jmattsson, 2015-07-02:
@shansted We've finally worked out why the filesystem doesn't work on the ESP-12E. There's a patch to the nodemcu esptool.py here, together with the explanation:
https://github.com/nodemcu/nodemcu-firmware/pull/523

anakod, 2015-07-02:
Strange, but I think it working right for me without additional path (UDK esptool.exe version) on NodeMCU-v2 board with 4Mb flash.

tprochazka, 2015-07-02:
> >So, you should change it in the build script because there is uppercase

> Where?

Sorry, you already changed it.

tprochazka, 2015-07-02:
I tried it again with latest version and I got this

```
ärlfs.start: size:3856 Kb, offset:0x38000

First init file system
sect_first: 38, sect_last: 3fc

5% 
...
103% 
104% 
105% 
formated
mount res: 0

Sming. Let's do smart things!
[W]sec 3fe error
....
[W]sec 3fe error
[W]sec 3fe error
[W]sec 3fe er
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 23344, room 16 
tail 0
chksum 0x18
load 0x3ffe8000, len 1352, room 8 
tail 0
chksum 0x2c
load 0x3ffe8550, len 3112, room 8 
tail 0
chksum 0x2e
csum 0x2e
Ś[02]ŐMEM CHECK FAIL!!!
ärl
```

Also is interesting that it initialize more than 100% of FS
I tested HttpServer_ConfigNetwork again.
It looks that first debug output "Sming. Let's do smart things!" works.

Squonk42, 2015-07-02:
> 100% is strange!

I did not check again with latest, but mine formatted only 100%, but I got "[W]sec 3fd error" instead of "[W]sec 3fe error" like you do.

I think there is something seriously broken with the Flash partitions, but I don't know what the layout is supposed to be on 512KB and 4MB chips.

shansted, 2015-07-02:
I was getting a similar error (Wsec error). Atleast in my setup, the post above from jmattsson was the cause. Updating esptool to latest version from the nodeMCU dev tree. 

The post in the link basically stated that when flashing in DIO mode, the flash was set read only on completion.

I've had my ESP12E's going since the initial 12E commit, as long a I flashed with the espressif tool. The update to the esptool has fixed this for me.

alon24, 2015-07-04:
so can we have the fixed esptool (exe) I do not know how to change sming to run the py script (it uses the exe i think on windows), I took latest and got the Wsec errors, and wdt reset every 5 seconds (in my code and in sming example configNetwork), so I do not know what the cause is but people say that the nodemcu tool took care of the issue, so I want it. help?

shansted, 2015-07-04:
Atleast on mine, in the same directory as the esptool, there is py2exe_esptool. This will generate a new exe.

Squonk42, 2015-07-05:
The link from @jmattsson above fixed the Wsec error.

Using this patch (https://github.com/nodemcu/nodemcu-firmware/commit/cfe5b23b476cea5036fbada09f16d5f76f992ab6), the filesystems can now be formatted upon first init, and example Basic_WiFi is flashing, and despite the filesystem is formatted up to 105% (!!!), he example is now working as expected, thanks!

Squonk42, 2015-07-05:
Digging further with the HttpSserver_AJAX example, I am asked to download some files by FTP th efirst time I access the module's Web page:

```Please connect to FTP and upload files from folder 'web/build' (details in code)```

This is exactly what I do using the command line Linux ftp client (advice: don't forget to force "binary" mode!), but the embedded FTP server seems partially broken, as I can only transfer 1 file at a time, and nned to disconnect and reconnect, or I get empty files for all subsequent transfers. Once a file is transferred, even a simple "ls" will not work correctly, it looks like the server is stuck in an intermediate state.

Can someone tell me if this issue Is this problem specific to NodeMCU-v2, or is it also the same using ESP-01 boards as well?

BotoX, 2015-07-05:
Hi, I'd like to share my experiences too.

I've been running the latest NodeMCU firmware on my NodeMCU devkit v1.0 and decided to switch over to Sming.
I am running Archlinux and it was easy to follow the instructions for linux, though there certainly is a lack of documentation - for example I was struggling with how to flash the generated files to my µC. (trial and error --> make flash)
I have tried out a few examples, the second one being the HttpServer_AJAX example which worked fine. Though I had to modify the makefile to use dio and 4M flash, also had use the patched esptool.py.

After the HttpServer_AJAX example I've tried out HttpServer_WebSockets which constantly gave me the following message to the console:
> no PleaseEnterSSID found, reconnect after 1s

I've confirmed that the WifiStation.config() function got the correct data - which it did. I deleted everything (Sming and esp-open-sdk) and tried again - same thing.

Then I flashed the nodemcu firmware again, authenticated with my WiFi network and flashed the HttpServer_WebSockets example afterwards. This time it successfully connected to my network and everything was working as it should.

This is the line I am using to flash my NodeMCU Devkit v1 (ESP-12E)
> /usr/bin/esptool.py -p /dev/ttyUSB0 -b 230400 write_flash -ff 40m -fm dio -fs 32m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x40000 out/firmware/spiff_rom.bin

Anyone with similar experiences or an explanation on this strange behaviour?

kireevco, 2015-07-05:
@BotoX I'd recommend joining us with [docker](https://github.com/kireevco/sming-docker) we try to unite our forces for a better cross-platform development environment.

tprochazka, 2015-07-14:
So, it's solved already? Ot the solution doesn't work work. Can somebody share fixed espotool.exe?

anakod, 2015-07-14:
Strange, but in my case original espotool works right..

kireevco, 2015-07-27:
@tprochazka I can suggest use a docker container that has an updated version of esptool.

tprochazka, 2015-08-23:
docker looks too complicated.

I simply installed python 2
Downloaded esptool.py from https://github.com/themadinventor/esptool
Changed path in Makefile-windows to
 ```ESPTOOL		 ?= py $(SDK_TOOLS)/esptool.py ```
And Nodemcu 1.0 now works correctly

It is possible to compile exe file by
 ```py esptool-build.py py2exe ```

Chocolatey package for sming is base on this SDK
https://github.com/CHERTS/esp8266-devkit
right?
There is still too old esptool implementation.

tprochazka, 2015-08-25:
Something i still not good for DIO mode programming od NodeMcu
I tried to play with HttpServer_ConfigNetwork it works correctly.
And then I flashed Basic_Blink to the NodemcuBoard.
LED is flashing, but i also write to serial output this

```
mode : sta(18:fe:34:a6:40:db) + softAP(1a:fe:34:a6:40:db)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with ATomAP, channel 3
dhcp client start...
ip:192.168.1.160,mask:255.255.255.0,gw:192.168.1.1
```

But Blink demo doesn't have Serial.begin, did not enable WIFI AP and last it still remember password to my home AP (I entered it during play with HttpServer_ConfigNetwork )

It mean that working code was preserver from previous flashed code?

anakod, 2015-08-27:
It is right, no problems here. Espressif SDK saving your WiFi parameters and apply it after reflashing.
It isn't related to flash size questions.

tprochazka, 2015-08-30:
Very interesting.
But what if I don't want to connect to AP, it can connect automatically without WifiStation.enable(true);
Maybe I can call WifiStation.enable(false);.

anakod, 2015-08-31:
Yes, you can of course, why not? :)

[3 more replies]

closed 2015-10-03 by hreintke