SmingHub/Sming #79: Autodiscover: mDNS classes

slaff, 2015-05-21:
Sming is a great framework for ESP8266 and is becoming my number one choice for development. What I am missing though are classes that can help the auto-discovery and zero configuration. As a start the possibility to create mDNS client would be great. This way the devices can discover and communicate easier with one another.

Example: New ESP device is connected to my local network. In the network there is a MQTT server and mDNS server that broadcasts the ip and port of the MQTT server. ESP reads the mDNS information about the MQTT server and connects directly without the need for extra configuration from a human.

I looked a bit and may be the code from this project https://github.com/sandeepmistry/esp8266-Arduino/tree/master/esp8266com/esp8266/libraries/ESP8266mDNS can be used as a start.

slaff, 2015-05-21:
See also https://github.com/anakod/Sming/issues/37

lopelex, 2015-05-21:
or use esp sdk
```cpp	
struct mdns_info *info = (struct mdns_info *)os_zalloc(sizeof(struct mdns_info));
info->host_name = “espressif”;
info->ipAddr = station_ipconfig.ip.addr; //ESP8266 station IP
info->server_name = “iot”;
info->server_port = 8080;
info->txt_data = “version = now”;
espconn_mdns_init(info);
```

hrsavla, 2015-05-25:
I wrote this small code based on Espressif SDK API. It is working. Typing "test.local" in browser will translate to ip address of your module. It needs to be tested if two ESP modules can resolve their address with mDNS.  (you need bonjour installed in Windows. Windows does not support mDNS without it. ). I haven't found way to work it on Android. Also test on Mac os and linux is to be done. 
void startmdns()
{
        struct mdns_info *info = (struct mdns_info *)os_zalloc(sizeof(struct
        mdns_info));
        info->host_name = (char *) "test";  // You can replace test with your own host name
        info->ipAddr = WifiStation.getIP(); 
        info->server_name = (char *) "iot";
        info->server_port = 80;
        char *xyz = (char *) "version = now";  // I still do not understand usage of TXT_data
        info->txt_data = xyz;

        espconn_mdns_init(info);
        espconn_mdns_server_register();
        espconn_mdns_enable();
}

slaff, 2015-05-25:
The thing is I want to be able to read information about advertised services like MQTT that are already provided from other devices on the network. I do not want the ESP to advertise something but to be able to read that mDNS service information. Something in the lines of:

```java
/**
     * Returns a list of service infos of the specified type.
     *
     * @param type
     *            Service type name, such as <code>_http._tcp.local.</code>.
     * @param timeout
     *            timeout in milliseconds. Typical timeout should be 6s.
     * @return An array of service instance.
     * @see https://svn.code.sf.net/p/jmdns/code/tags/jmdns-3.4.1/src/main/java/javax/jmdns/JmmDNS.java
     */
    public ServiceInfo[] list(String type, long timeout);
```

So that I can say
```c
serviceList = mdns.list('_mqtt._tcp');
```
and I can iterate over the list and find the port, ip and extra data stored as key-value.

hrsavla, 2015-05-26:
You are correct ESP8266 API just does Advertise. Listen feature will also be needed for ESP8266 to ESP8266 communication.
TI cc3200 SDK.1.1 has this facility. It would be nice to have it for ESP8266.

http://gkaindl.com/software/arduino-ethernet/bonjour
Above code seems to have this Listen function. You can try port it into Sming. I will try to help but first I will need 2 ESP modules. Or I have to figure out way to start mDNS advertiser on my windows Pc.

tprochazka, 2015-06-14:
On Android you can try this
https://play.google.com/store/apps/details?id=com.grokkt.android.bonjour
From implementation point of view you can try
http://jmdns.sourceforge.net/ which should work on Android
or native
http://developer.android.com/training/connect-devices-wirelessly/nsd.html
But I didn't try it personally.

But I'm looking now for the ways how to localise my ESP8266 device in home networks.

tprochazka, 2015-06-14:
@hrsavla What I must do to compile your startmdns function? It doesn't work in my Sming project, some include is necessary?

hrsavla, 2015-06-14:
It should work on simg .  I have tested it. Code is based on esp skd. Kindly first check mdns works on your pc/laptop/phone. On Android decoding will work app u told but browser in android does not translate eg. Test.local to ip address. In pc translation is done on browser.

hrsavla, 2015-06-14:
Esp to esp ipaddress translation with mdns is not supported in above code.

tprochazka, 2015-06-14:
Its not possible to compile it.

I tried
```
void startmdns() {
	struct mdns_info *info = (struct mdns_info *)os_zalloc(sizeof(struct mdns_info));
	info->host_name = (char *) "test"; // You can replace test with your own host name
	info->ipAddr = WifiStation.getIP();
	info->server_name = (char *) "iot";
	info->server_port = 80;
	char *xyz = (char *) "version = now"; // I still do not understand usage of TXT_data
	info->txt_data = xyz;

    espconn_mdns_init(info);
    espconn_mdns_server_register();
    espconn_mdns_enable();
}
```

and it fails with
```
C+ app/SmartESP8266.cpp
app/SmartESP8266.cpp: In function 'void startmdns()':
app/SmartESP8266.cpp:130:80: error: invalid application of 'sizeof' to incomplete type 'startmdns()::mdns_info'
  struct mdns_info *info = (struct mdns_info *)os_zalloc(sizeof(struct mdns_info));
                                                                                ^
app/SmartESP8266.cpp:131:6: error: invalid use of incomplete type 'struct startmdns()::mdns_info'
  info->host_name = (char *) "test"; // You can replace test with your own host name
      ^
app/SmartESP8266.cpp:130:9: error: forward declaration of 'struct startmdns()::mdns_info'
  struct mdns_info *info = (struct mdns_info *)os_zalloc(sizeof(struct mdns_info));
         ^
app/SmartESP8266.cpp:132:6: error: invalid use of incomplete type 'struct startmdns()::mdns_info'
  info->ipAddr = WifiStation.getIP();
      ^
app/SmartESP8266.cpp:130:9: error: forward declaration of 'struct startmdns()::mdns_info'
  struct mdns_info *info = (struct mdns_info *)os_zalloc(sizeof(struct mdns_info));
         ^
app/SmartESP8266.cpp:133:6: error: invalid use of incomplete type 'struct startmdns()::mdns_info'
  info->server_name = (char *) "iot";
      ^
app/SmartESP8266.cpp:130:9: error: forward declaration of 'struct startmdns()::mdns_info'
  struct mdns_info *info = (struct mdns_info *)os_zalloc(sizeof(struct mdns_info));
         ^
app/SmartESP8266.cpp:134:6: error: invalid use of incomplete type 'struct startmdns()::mdns_info'
  info->server_port = 80;
      ^
app/SmartESP8266.cpp:130:9: error: forward declaration of 'struct startmdns()::mdns_info'
  struct mdns_info *info = (struct mdns_info *)os_zalloc(sizeof(struct mdns_info));
         ^
app/SmartESP8266.cpp:136:6: error: invalid use of incomplete type 'struct startmdns()::mdns_info'
  info->txt_data = xyz;
      ^
app/SmartESP8266.cpp:130:9: error: forward declaration of 'struct startmdns()::mdns_info'
  struct mdns_info *info = (struct mdns_info *)os_zalloc(sizeof(struct mdns_info));
         ^
app/SmartESP8266.cpp:138:27: error: 'espconn_mdns_init' was not declared in this scope
     espconn_mdns_init(info);
                           ^
app/SmartESP8266.cpp:139:34: error: 'espconn_mdns_server_register' was not declared in this scope
     espconn_mdns_server_register();
                                  ^
app/SmartESP8266.cpp:140:25: error: 'espconn_mdns_enable' was not declared in this scope
     espconn_mdns_enable();
                         ^
make: *** [out/build/app/SmartESP8266.o] Error 1
```

hrsavla, 2015-06-14:
Sdk is not in your compiler path. Kindly check C:\espressif \sdk . I had last checked on sdk 1.01. I will confirm for sdk 1.12 in evening.

tprochazka, 2015-06-14:
I'm using default sming makefile that is used for all sample apps.
I hoped that it include espessif SDK by default, it should not work without it.
c:\Espressif\sdk\ exist on my system.
I will check more deeply how it works.

tprochazka, 2015-06-14:
So I had some not final version of 1.0.1 where mdns_info  was missing.
Now it works correctly.
I tested it with this app https://play.google.com/store/apps/details?id=com.grokkt.android.bonjour
And it found _iot._tcp on local domain, when I open it sometimes really found 'test'

But most time it crash

```
Fatal exception (28):
epc1=0x402247a6, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
rm match
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

tprochazka, 2015-06-14:
Anw SDK 1.1.0 is not supported by sming yet.
```
d:/_Elektro/ESP8266/Espressif/sdk/lib\libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x34c): undefined reference to `user_rf_pre_init'
c:/Espressif/sdk/lib\libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x474): undefined reference to `user_rf_pre_init'
collect2.exe: error: ld returned 1 exit status
```

hrsavla, 2015-06-14:
Yes with android app espressif crashes. Reason is not known. I had put a query to bbs.espressif for this about a month back. But no reply from them. It works with pc (you have to install bonjour). 
Solutions to make latest sdk 1.1.2 to work is given by alonewolfx2 in his PR.

tprochazka, 2015-06-14:
Thanks a lot, PR by alonewolfx2 helps.
Can you provide link to the BBS with you bug report?

hrsavla, 2015-06-14:
http://bbs.espressif.com/viewtopic.php?f=7&t=485&p=0&e=0&sid=61fdd584db40cd3a4d3216d79fde5dc9

Above is the link

tprochazka, 2015-06-14:
Thanks.

I found, if I skip call
espconn_mdns_server_register();
espconn_mdns_enable();
so only  espconn_mdns_init(info); will remain.
It doesn't crash.

Bonjour Browser doesn't found any running service on IoT, but for example this one
https://play.google.com/store/apps/details?id=com.melloware.zeroconf
found everything.

hrsavla, 2015-06-15:
Advertise and listening are two functions of mdns. Code implemented with esp sdk only does advertise. For listening you might have to port this bellow code 
http://gkaindl.com/software/arduino-ethernet/bonjour

slaff, 2015-06-15:
Thank you @hrsavla . ArduinoEthernet has exactly what I am looking for. It would be great if someone port the library to ESP8266/Sming.

closed 2015-07-02 by anakod