SmingHub/Sming #84: Makefile-windows.mk problem

tprochazka, 2015-05-25:
I have problem with default makefile for windows on these lines:

```
GEN_APPBIN   := PATH="$(ESP_HOME)/xtensa-lx106-elf/bin:$(PATH)" && $(SDK_TOOLS)/gen_appbin.exe
GEN_FLASHBIN := PATH="$(ESP_HOME)/xtensa-lx106-elf/bin:$(PATH)" && $(SDK_TOOLS)/gen_flashbin.exe
```

Because PATH on window contain space and also : is not right path delimiter on windows.
So I completely removed :$(PATH) part and it now works correctly

So it now looks like
```
GEN_APPBIN   := PATH="$(ESP_HOME)/xtensa-lx106-elf/bin" && $(SDK_TOOLS)/gen_appbin.exe
GEN_FLASHBIN := PATH="$(ESP_HOME)/xtensa-lx106-elf/bin" && $(SDK_TOOLS)/gen_flashbin.exe
```

kireevco, 2015-05-25:
This way you will be ignoring all other contents of your PATH, which is not right.
What exact error did you see?

tprochazka, 2015-05-26:
This command doesn't need any other locations.
I will send you exact error message later.

tprochazka, 2015-05-27:
Sorry for delay, here is build output from eclipse with original makefile

```
C+ /d/_Elektro/ESP8266/sming/Sming/appinit/user_main.cpp
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out
Running objcopy, please wait...
Objcopy done
Run gen_appbin
/bin/sh: Files/Java/jdk1.8.0_40/bin:/d/_PJazyky/_versioning/Git/cmd:/d/_PJazyky/_versioning/Git/bin:/d/_PJazyky/Python:/c/Program Files (x86)/proXPN/bin:/c/Program Files (x86)/Common Files/Teleca Shared:/c/Program Files (x86)/Skype/Phone/:/d/_PJazyky/MinGW/bin:/usr/bin:/c/Program Files/Ruby200-x64/bin: No such file or directory
make: *** [out/build/app.out] Error 127
```

Without :$(PATH) everything works correctly.

RibeiroSantos, 2015-06-02:
Hi,

Same error, resolve adding ";" after $(PATH)

GEN_APPBIN   := PATH="$(ESP_HOME)/xtensa-lx106-elf/bin:$(PATH);" && $(SDK_TOOLS)/gen_appbin.exe
GEN_FLASHBIN := PATH="$(ESP_HOME)/xtensa-lx106-elf/bin:$(PATH);" && $(SDK_TOOLS)/gen_flashbin.exe

tprochazka, 2015-06-13:
It doesn't works for me (from Eclipse)
Onyl if I complete remove :$(PATH)

kireevco, 2015-06-13:
Can you try removing quotes and keep the original line?

kireevco, 2015-06-13:
This is a problem apparently with gen_appbin. Actual build is working fine.

anakod, 2015-06-13:
Why PATH is needed if we already specified correctly full path?

kireevco, 2015-06-13:
$(ESP_HOME)/xtensa-lx106-elf/bin is required for app_gen mostly.
We can try to remove that dependency by using modern esptool that might have all required functionality that is used from app_gen

kireevco, 2015-06-28:
@tprochazka I updated makefile. Please let us know how it works with the latest Makefile-windows.mk

closed 2015-07-27 by kireevco

tprochazka, 2015-11-03:
I confirm that it works now. Thanks