SmingHub/Sming #86: Compilation issues with SDK 1.1

slaff, 2015-05-25:
During compilation with the latest version of the esp-open-sdk (https://github.com/pfalcon/esp-open-sdk/) using Espressif SDK 1.1 I got the following error:

```bash
C+ SmingCore/Digital.cpp
SmingCore/Digital.cpp: In function 'void pullup(uint16_t)':
SmingCore/Digital.cpp:78:43: error: 'PIN_PULLDWN_DIS' was not declared in this scope
  PIN_PULLDWN_DIS((EspDigitalPins[pin].mux));
                                           ^
SmingCore/Digital.cpp: In function 'void pulldown(uint16_t)':
SmingCore/Digital.cpp:86:42: error: 'PIN_PULLDWN_EN' was not declared in this scope
  PIN_PULLDWN_EN((EspDigitalPins[pin].mux));
                                          ^
SmingCore/Digital.cpp: In function 'void noPullup(uint16_t)':
SmingCore/Digital.cpp:92:43: error: 'PIN_PULLDWN_DIS' was not declared in this scope
  PIN_PULLDWN_DIS((EspDigitalPins[pin].mux));
                                           ^
make: *** [out/build/SmingCore/Digital.o] Error 1
```

slaff, 2015-05-25:
That link here can point out the issue: http://bbs.espressif.com/viewtopic.php?f=7&t=494

anakod, 2015-06-04:
Please check it.

slaff, 2015-06-04:
Works like a charm :)

closed 2015-06-04 by slaff