SmingHub/Sming #99: Feature-Request: HTTPS support in HttpClient

slaff, 2015-05-29:
It looks like the HttpClient supports only http.

```CPP
bool HttpClient::startDownload(URL uri, HttpClientMode mode, HttpClientCompletedCallback onCompleted)
{
	//...
	if (uri.Protocol != "http") return false;
```
It would be very helpful to have also HTTPS support.

anakod, 2015-06-04:
I think it should be switchable option, because HTTPS is very expensive in memory usage

antelder, 2015-06-16:
+1 to https. all over the net people are looking for https support, would be a killer feature for sming. I just need a single https post to get an auth token...

tprochazka, 2015-06-18:
I saw that current Espressif SDK already has direct support for HTTPS

anakod, 2015-06-18:
Looks like SDK doesn't have correct HTTPS verification: http://www.esp8266.com/viewtopic.php?f=6&t=3343&p=19188&hilit=sha1#p19188

robotiko, 2015-07-06:
Maybe this can be an starting point:
and ESP project that implements https 
https://github.com/SuperHouse/esp-open-rtos

ported lib 
 https://github.com/SuperHouse/axtls as a submodule optimized for low memory devices.

slaff, 2015-09-16:
The Arduino/ESP project already has HTTPS support ( https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp ). May be someone with good C/C++ skills can port this code to Sming?

harry-boe, 2016-03-14:
bump for this issue.
 I guess we can close this with the new SSL implementation #596

closed 2016-03-14 by hreintke

ADiea, 2016-03-14:
Hi but does the http client open a HTTP conn or a HTTPS connection to the server based on the protocol part in the URL? Sorry diodn't have time to test the new SSL implementation