SmingHub/Sming #102: WS2812 after 6 changes reboot

ystrem, 2015-05-30:
Hi, I'm testing the WS2812 example and after 6 changes of color I got reboot. I have ESP12 and using 1metr long LED strip. On Arduino Uno it's ok, so the strip is OK.

anakod, 2015-06-04:
What it write when reboot?
May be you should disable\feed WDT?
```C++
WDT.enable(false);
```

kireevco, 2015-06-23:
@ystrem can you please report back whenever you can with WDT disabled?

kireevco, 2015-07-27:
@ystrem any update?

dounix, 2015-10-03:
I get the same as @ystrem, a reboot after about 6 seconds(I don't think the transition count matters)
Tried with WDT.enable(false); , same results.
This fails for me on both an esp03(4Mbit flash) and esp12e(32Mbit flash)
###### Things I tried that made no difference
* WDT disabled
* modifying the delay time in the demo
* remove the 2nd blink, just a single color pushed in the loop(with the delay)

###### Console output during reboot
     ets Jan  8 2013,rst cause:2, boot mode:(3,0)
    
    load 0x40100000, len 20372, room 16 
    tail 4
    chksum 0xb6
    load 0x3ffe8000, len 848, room 4 
    tail 12
    chksum 0xa4
    ho 0 tail 12 room 4
    load 0x3ffe8350, len 1292, room 12 
    tail 0
    chksum 0x8b
    csum 0x8b
     rtc mem data
    rlfs.start: size:3856 Kb, offset:0x0x38000
    
    mount res: 0

I'm willing to test some things out if there are any suggestions

dounix, 2015-10-03:
This can be reproduced in it's simplest form with this minimal code
```
#include <user_config.h>
#include "SmingCore/Platform/WDT.h"
void init()
{
    WDT.enable(false);  //doesn't matter..
    while (true) {
       //matters not what you put here, it will crash
    }
}
```
I have a PR that fixes the example(with timers), but not the while loop induced crash.

alonewolfx2, 2015-10-03:
@dounix Please pull PR into the develop.
@anakod @kireevco basically wdt class not working correctly with new sdk 1.3. i will change class and make  PR.

closed 2015-10-03 by alonewolfx2