SmingHub/Sming #106: ets_vsnprintf

hreintke, 2015-05-31:
Anakod,
In your microc lib I see references to : 
```
int ets_vsprintf(char *str, const char *format, va_list argptr);
int ets_vsnprintf(char *buffer, size_t sizeOfBuffer, const char *format, va_list argptr);
```
But if I use these functions directly in my sming environment I get the linker error : 
```
 undefined reference to `ets_vsprintf(char*, char const*, __va_list_tag)
```
Do you know where this functions are located ? Do I need to include another library ?

hreintke, 2015-05-31:
Anakod,
Just found (maybe) the way how ik works. I'll check that first.
Don't know how to delete the issue so I closed it.

closed 2015-05-31 by hreintke

anakod, 2015-06-01:
Main feature what it's using internal Espressif ROM and doesn't requrie additional code (and program space).

hreintke, 2015-06-01:
Yes, That's what I expected but I can't get it used.
I added the following to HardwareSerial for a proof of concept of printf : 
```
void HardwareSerial::printf(const char *fmt, ...)
{
	char x [256];
	size_t sz;
	// Doesn't work :(
	va_list va;
	va_start(va, fmt);
//	ets_uart_printf(fmt, va);
//	sz = vsnprintf(x,255, fmt, va);
	sz = vsprintf(x,fmt,va);
	va_end(va);
	for (int i=0;i<sz;i++){write(x[i]);};
}```
This ony works when I use the vsprintf, when using vsnprintf or ets_vsprintf, ets_vsnprintf. I can compile the framework but get link errors when compiling/linking the example applications.
I did use
```
#include <stdio.h>
```
But are not sure which one is then included 
I see in libmicroc.a both vsnprintf.and vsprintf.o, so I expected them to be available, as also the ets_ versions of them in the sdk.