SmingHub/Sming #107: IPAddress String usage

hreintke, 2015-06-02:
I am working on my TelnetServer implementation and one of my debugf statements is : 
```
bool TelnetServer::telnetReceive (TcpClient& client, char *data, int size)
{
	debugf("TelnetServer callback : %s, %d bytes \r\n",client.getRemoteIp().toString().c_str(),size );
```
When compiling the sming framework with this I get the error : 
```
Wiring/IPAddress.h:26:7: error: forward declaration of 'class String'
 class String;
       ^
```
Updating in IPAdress.h : 
```
//class String;
#include "WString.h"
```
I can compile without errors.
Do you expect that this update will not influence other use of IPAddress of should I look for another solution.

anakod, 2015-06-03:
You can include WString.h in your file, of course, but I think what this fix will be good to make usage more simple.

anakod, 2015-06-03:
May be you will make P-R?

hreintke, 2015-06-05:
OK, I will submit P-R

closed 2015-06-09 by hreintke