SmingHub/Sming #108: Add servo lib support

robotiko, 2015-06-03:
Servo seems to be one of those Arduino AVR libs that doens't work due to highly coupled AVR timming. 
There is a new port to ESP8266 arduino in the main repo: 

https://github.com/esp8266/Arduino/tree/esp8266-sdk-1.0/hardware/esp8266com/esp8266/libraries/Servo

kireevco, 2015-06-23:
@robotiko PRs are welcome :)

robotiko, 2015-06-24:
Still dealing with newbie issues, but I will try to address that as small contribution to this great project as soon as I know how it goes. 
:)

hreintke, 2015-11-14:
@robotiko : 
Are you still working on this or can we close the issue ?

robotiko, 2015-11-14:
I have a single servo working, however it suffers from not precise timing and it is not really usable until HW timer is available and stable. 
We can close an reopen later.

hreintke, 2016-01-27:
Solved

closed 2016-01-27 by hreintke