SmingHub/Sming #119: Problem Build !

falbalala, 2015-06-14:
:rage::question:
***
01:21:39 **** Build of configuration Sming for project LED_WS2812-LAST ****
make all 
Building file: ../app/application.cpp
Invoking: GCC C++ Compiler
g++ -I"c:\tools\Sming\Sming" -I"c:\tools\Sming\Sming/system/include" -I"c:\tools\Sming\Sming/Libraries" -I"c:\Espressif/sdk/include" -O2 -g -Wall -c -fmessage-length=0 -MMD -MP -MF"app/application.d" -MT"app/application.d" -o "app/application.o" "../app/application.cpp"
../app/application.cpp:1:25: fatal error: user_config.h: No such file or directory
 #include <user_config.h>
                         ^
compilation terminated.
make: *** [app/application.o] Error 1
***

Thx in advance :zzz:

kireevco, 2015-06-23:
@falbalala Please provide us some more information. How did you install sming? Is any other building / was building?

kireevco, 2015-06-28:
@falbalala Windows README was updated. Please make sure to use -source parameter.
And let me know if you still have any issues.

Also, consider using https://github.com/kireevco/sming-docker with NetBeans. Much Better!

closed 2015-07-27 by kireevco