SmingHub/Sming #122: Issues with windows install

robotiko, 2015-06-17:
Hi, 
First of all, thank you for such great project and all the hard work on it. 
I just tried to install in a win32 win7 environment following the nice Wiki Windows-Quickstart but I found several issues with the install. 

1) Just installing with the recommended full install, but it looks like didn't install the Unofficial Development Kit. 
I went ahead and installed manually (latest version)  and then installed examples with choco .. but none of the projects build with the same errors as in  #86.

Tried to update using choco  and got the message "sming v1.0.5.20150507 is the latest version available based on your source(s)"
Obviously this is not the latest version. 
How to proceed?

I tried to uninstall using choco  to make a clean install, but it seems to leave 1G orphan files and eclipse, etc so there is no uninstall available I guess. 

Any advice on how to get latest version running?  

Thank you

anakod, 2015-06-18:
You also can manually update Sming sources from GitHub (download as ZIP)

robotiko, 2015-06-18:
Thanks @anakod, 
Updated the Unofficial Development Kit and then sming and now seems to work nicely. 

From a clean new entry to sming point of view,  Windows Wiki, choco releases seem to be outdated and affected by the initial sdk 1.1 issues. 
From my side I would close the issue, however the issue when follow the wiki is still there so it also makes sense to leave it open for tracking.

After seeing the current sming-docker status,  it looks much more isolated, easier to access and very promising.  
Thank for such great stuff!

kireevco, 2015-06-23:
@robotiko can you please reiterate, what is the issue you're facing eventually? Is it 'newer version missing' or some additional problems?

Thank you

robotiko, 2015-06-24:
@kireevco, 
there are several issues related: #86, #126 and I bet that #119 can also be related. 

The problem I had  is that just clean install following the wiki for a frictionless choco install didn't work out of the box.  
1) It didn't install the  Unofficial Development Kit. 
2) With a manual install (when I did it was previous version to current  v2.0.6) that got solved, I installed examples with choco, but were failing to build with the same errors as #86 (v1.1.0 stuff )
3) Tried to update sming and the examples using choco, but it seem that it just keep a quite aged version (month old).
4) I got also hit by path issue #84 

Finally once I figured out "where choco leaves what", I went ahead with a git update of the sming folder   (core + examples) and got a just released 2.0.6 Unofficial Development Kit an everything started to work. 

One nasty thing about the choco is that uninstall doesn't work properly.  I'm not  sure what it does, if it does something at all, but at least eclipse, SDK, environment  (and not sure about sming) are left in hd untouched so a clean retry is not possible. 

As I mentioned, I fixed all issues by reading the repo issues and the manual install, but was far from install choco and type install. :)

I hope it helps.

kireevco, 2015-06-28:
@robotiko Windows README was updated. please make sure to use -source parameter.
Please let me know if you still have any issues.

Also, consider using https://github.com/kireevco/sming-docker with NetBeans. Much Better!

closed 2015-06-28 by kireevco

GavinAndrews, 2015-11-10:
Hi

I am trying to follow the instructions on the Windows Quick Start page

I cannot install the examples...

```
 Get-BinRoot is going to be deprecated by v1. Many packages no longer require it
 since the folders no longer have versions on them.
 Join-Path : Cannot bind argument to parameter 'Path' because it is null.
 At C:\ProgramData\chocolatey\lib\sming.examples\tools\chocolateyInstall.ps1:7 c

 har:30
 + $eclipseWorkspace = Join-Path <<<<  $env:HOMEPATH "sming.examples"
     + CategoryInfo          : InvalidData: (:) [Join-Path], ParameterBindingVa

    lidationException
     + FullyQualifiedErrorId : ParameterArgumentValidationErrorNullNotAllowed,M

    icrosoft.PowerShell.Commands.JoinPathCommand
```

in red text to start... then downloads and extracts... but then hangs.

Any ideas?