SmingHub/Sming #126: choco install sming -y failed

hwiguna, 2015-06-19:
I followed the instructions for installing sming using chocolatey.  All went well until this command:
choco install sming -y

It failed with this error.  The log basically contains the same error.

 Get-BinRoot is going to be deprecated by v1. Many packages no longer require it
 since the folders no longer have versions on them.
 Installing / updading required mingw packages
 Exception calling "Start" with "1" argument(s): "The system cannot find the fil
e specified"
 At C:\ProgramData\chocolatey\helpers\functions\Start-ChocolateyProcessAsAdmin.p
s1:76 char:3
 +   $s = [System.Diagnostics.Process]::Start($psi)
 +   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException
     + FullyQualifiedErrorId : Win32Exception
 You cannot call a method on a null-valued expression.
 At C:\ProgramData\chocolatey\helpers\functions\Start-ChocolateyProcessAsAdmin.p
s1:81 char:3
 +   $s.StandardError.ReadToEnd() | Out-File $errorFile
 +   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     + CategoryInfo          : InvalidOperation: (:) [], RuntimeException
     + FullyQualifiedErrorId : InvokeMethodOnNull
 You cannot call a method on a null-valued expression.
 At C:\ProgramData\chocolatey\helpers\functions\Start-ChocolateyProcessAsAdmin.p
s1:82 char:3
 +   $s.WaitForExit()
 +   ~~~~~~~~~~~~~~~~
     + CategoryInfo          : InvalidOperation: (:) [], RuntimeException
     + FullyQualifiedErrorId : InvokeMethodOnNull
 Get-Content : Cannot find path 'C:\Users\hwiguna\AppData\Local\Temp\chocolatey\
-error.stream' because it does not exist
 .
 At C:\ProgramData\chocolatey\helpers\functions\Start-ChocolateyProcessAsAdmin.p
s1:88 char:21
 +       $innerError = Get-Content $errorFile | Out-String
 +                     ~~~~~~~~~~~~~~~~~~~~~~
     + CategoryInfo          : ObjectNotFound: (C:\Users\hwigun...y\-error.strea
m:String) [Get-Content], ItemNotFoundEx
    ception
     + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.GetCon
tentCommand
 [ERROR] Running C:\tools\mingw64\bin\mingw-get.exe with install mingw32-base --
reinstall --recursive was not
 successful. Exit code was '' Error Message: .
 At C:\ProgramData\chocolatey\helpers\functions\Start-ChocolateyProcessAsAdmin.p
s1:92 char:5
 +     throw $errorMessage
 +     ~~~~~~~~~~~~~~~~~~~
     + CategoryInfo          : OperationStopped: ([ERROR] Running...rror Message
: .:String) [], RuntimeException
     + FullyQualifiedErrorId : [ERROR] Running C:\tools\mingw64\bin\mingw-get.ex
e with install mingw32-base --reinstall
     --recursive was not successful. Exit code was '' Error Message: .
The install of sming was NOT successful.
Error while running 'C:\ProgramData\chocolatey\lib\sming\tools\chocolateyInstall
.ps1'.
 See log for details.

kireevco, 2015-06-23:
@hwiguna I'm checking something. Meanwhile, can you please make sure you are running as an admin. you can also try installing mingw separately: `choco install mingw`

kireevco, 2015-06-28:
@hwiguna Windows README was updated. please make sure to use -source parameter.
Please let me know if you still have any issues.

closed 2015-06-28 by kireevco

avsokolov, 2016-01-20:
I followed the updated instructions for installing sming using chocolatey. All went well until this command:
`choco install sming.core -y -source 'https://www.myget.org/F/kireevco-chocolatey/'`

But it still failed with error _**https://github.com/anakod/Sming/archive/2.1.0.zip - 404**_ 
Details:
```
Installing the following packages:
sming.core
By installing you accept licenses for the packages.

sming.core v2.1.0
 Get-BinRoot is going to be deprecated by v1. Many packages no longer require it since the folders no longer have versions on them.
 Attempt to get headers for https://github.com/anakod/Sming/archive/2.1.0.zip failed.
   Exception calling "GetResponse" with "0" argument(s): "The remote server returned an error: (404) Not Found."
 Downloading sming 32 bit
   from 'https://github.com/anakod/Sming/archive/2.1.0.zip'
 Exception calling "GetResponse" with "0" argument(s): "The remote server returned an error: (404) Not Found."
 At C:\ProgramData\chocolatey\helpers\functions\Get-WebFile.ps1:81 char:3
 +   $res = $req.GetResponse();
 +   ~~~~~~~~~~~~~~~~~~~~~~~~~
     + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException
     + FullyQualifiedErrorId : WebException
 You cannot call a method on a null-valued expression.
 At C:\ProgramData\chocolatey\helpers\functions\Get-WebFile.ps1:166 char:3
 +   $res.Close();
 +   ~~~~~~~~~~~~
     + CategoryInfo          : InvalidOperation: (:) [], RuntimeException
     + FullyQualifiedErrorId : InvokeMethodOnNull
 Chocolatey expected a file to be downloaded to 'C:\Users\alexsk\AppData\Local\Temp\chocolatey\sming\2.1.0\smingInstall.zip' but nothing exists at that location.
 C:\ProgramData\chocolatey\helpers\functions\Get-ChocolateyWebFile.ps1:158 знак:24
 +   if (!($fi.Exists)) { throw "Chocolatey expected a file to be downloaded to `'$ ...
 + ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     + CategoryInfo          : OperationStopped: (Chocolatey expe... that location.:String) [], RuntimeException
     + FullyQualifiedErrorId : Chocolatey expected a file to be downloaded to 'C:\Users\alexsk\AppData\Local\Temp\chocolatey\sming\2.1.0\smingInstall.zip' but nothing exists at that location.
The install of sming.core was NOT successful.
Error while running 'C:\ProgramData\chocolatey\lib\sming.core\tools\chocolateyInstall.ps1'.
 See log for details.

Chocolatey installed 0/1 package(s). 1 package(s) failed.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
Failures:
 - sming.core
```