SmingHub/Sming #127: Feature Request: Option to use RTOS-SDK

0xPIT, 2015-06-21:
I'd like to see Sming compiling with the https://github.com/espressif/esp_iot_rtos_sdk.

0xPIT, 2015-06-28:
I've forked it and made it compile, It's not yet working, though. Currently working on serial.

hreintke, 2015-06-28:
@0xPIT : What would be the benefit of using RTOS-SDK ?
The espressif sdk gets regular updates, has a lot of users and espressif replies with updates/patches to bugs. RTOS-SDK seems rather quiet.

0xPIT, 2015-06-28:
I think it's been updated a couple of weeks ago.

IMHO FreeRTOS saves quite some work (like queues that are interrupt save) and you can properly use tasks and structure code better.

kireevco, 2015-06-28:
+ please

ystrem, 2015-07-05:
That would be great :)

ystrem, 2015-07-05:
What do you think about this https://github.com/SuperHouse/esp-open-rtos ?

hreintke, 2015-11-14:
This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-11-14 by hreintke

malachib, 2016-12-16:
I'm thinking of bringing in bits at a time for my own projects using esp-open-rtos.  It seems the best approach would be to #ifdef relevant parts in a forked version of the Sming repo and pull request it back again, rather than the noble effort by @hreintke to maintain a completely separate repo.

pros:

* easier path to mature and up-to-date code

cons:

* uglifies some existing framework code with #ifdef's

I'd start by using esp-open-rtos as my foundation for makes, etc - probably leaving sming-centric makes to others.  

Is this something interesting to us?  Down the line maybe even toss in some CMSIS #ifdefs?  It only feels viable if the Sming community is interested in "mainlining" (love that term ha) these changes.

slaff, 2016-12-17:
> Is this something interesting to us?

@malachib That sounds great. Just make sure to target your efforts towards ESP32-IDF #831.

malachib, 2016-12-17:
@slaff that would be an ideal.  I conjecture ESP-IDF and esp-open-rtos bear far more similarities than differences and if so, combining and/or aligning such efforts makes a lot of sense.  Seems to me choosing and policing a branching/merging strategy is more important than usual for this scenario.

malachib, 2016-12-19:
Interesting relevant discussion here http://www.esp32.com/viewtopic.php?t=362