SmingHub/Sming #128: Problems with generating spiff files to flash (in example HttpServer_Websockets)

kennpie, 2015-06-21:
1) using CHERTS Unofficial Development Kit for Windows (v2.06), but ESP SDK 1.01b1, the make script can't find 'spiffy' to process content in /files. I found spiffy.exe in the NodeLua project, and moved it to various places, including the HttpServer_Websockets directory, but the make file never seemed to find it.

I found that if I commented out line 415  'spiffy;\' in C:\Espressif\sming\Sming\Makefile-project.mk, manually ran spiffy.exe in the project folder to build spiff_rom.bin, then ran make all or make rebuild again, make would find and move the manually-generated spiff_rom.bin to the out/firmware dir.

2) after flashing (a known-good  ESP-01) with the compiled bins, the running server can't find the files that should have been in the uploaded spiff_rom.bin. I did open spiff_rom.bin in an editor, and confirmed that contained the expected files... but of course I don't know if spiffy.exe is doing the right thing.

Thanks.

kireevco, 2015-06-23:
@kennpie can you please give us more detail, where did you get spiffy.exe? Where is it currently located?
Have you tried installing newer spiffy: ```choco install spiffy``` or from [here](https://bintray.com/kireevco/generic/spiffy/1.0.3/view) ?

kennpie, 2015-06-23:
Hi kireevco,
I found 'spiffy.exe' as part of an example in CHERTs SDK - C:\Espressif\examples\nodemcu-spiffy. It seems to work but the flashed ESP8266 couldn't read it.

I guess what I'm asking is:
- where to get the right spiffy (which you've just answered, thank you)
- where to put it.

So where should I install it in my Windows dev environment?

Thanks,

Ken
Toronto

kireevco, 2015-06-23:
@kennpie Spiffy from nodemcu is not compatible with sming (doesnt't work with large files and has some other bugs). You can put it anywhere in the PATH variable. If you used `choco install sming -y` it will be automatically put in c:\tools\mingw64\bin\spiffy.exe.

kennpie, 2015-06-24:
Thanks for the fast reply. (and for porting the spiffy) I put your spiffy.exe in the path, and it is now found and used during builds. So that's one problem solved.

However, at present, the ESP8266 apparently still can't read it. I'm using the HttpServer_Websockets example without changes and an ESP-01. The project builds, the spiff-rom.bin is generated, it gets flashed... but when running, the ESP doesn't seem to read it. I added a directory lister to the init, and it reports no files found. I just get 404 - Not Found when I hit it. Other examples that load themselves, like HttpServer_Bootstrap, have worked fine.

I'll keep trying.

Thanks,

Ken

kireevco, 2015-06-28:
@kennpie looks like an issue with HttpServer_Websockets.

@anakod ?

robotiko, 2015-07-05:
Hi @kennpie  Running latest version (1.2) in nodemcu dev v0.9 and working fine.  
I just added a filesystem lister in the "onIndex()" and added 10 files to the /files folder and all get listed there without issues. 
Where did you added your files?

kennpie, 2015-07-05:
Hi robotiko, and thanks for the report.

In my test, I added web content to the 'files' directory of the HttpServer_Websockets project, and in a hex editor I was able to confirm that the built spiff-rom.bin file contained those files.. and that that file was flashed... but the ESP8266 couldn't find the files.

I will try this again shortly.

ADiea, 2015-07-13:
+1 same thing happens to me. I put index.html in files folder builds fine flashes fine but the file cannot be found on runtime. Didn't have time to investigate, I will try using the spiffy listed here. 

@kennpie When booting up you should see this line in console

```
fs.start: size:224 Kb, offset:0x4400
```

It appears that the offset is wrong, I will investigate when I have some time, please tell us what you get in console.

kireevco, 2015-07-13:
Let's close this spiffy issue, and create a new one for websockets example?

ADiea, 2015-07-13:
I don't think this is a websockets issue this happened to me with ajax example also. I will investigate a little in a few hrs

robotiko, 2015-07-13:
@ADiea, @kennpie, 
 it might be a memory size/mapping thing. How much memory device are you using? 
is it 4MB version?

ADiea, 2015-07-13:
I'm using the 512KB flash. I put some debug prints in \Sming\Sming\Services\SpifFS/spiffs.c

```
spiffs: cfg.phys_addr:0x40240000
spiffs: INTERNAL_FLASH_SIZE:496 KB
spiffs: cfg.phys_size:245760
spiffs: cfg.phys_erase_block:4096
spiffs: cfg.log_block_size:8192

fs.start: size:240 KB, offset:0x40000
```

from linker script

```
irom0_0_seg :  org = 0x40209000, len = 0x42000
```

and when i compile my app

```
#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8582|    1410
    rodata|           ReadOnly Data (RAM)|    3FFE8590|    3FFEA1DC|    7244
       bss|      Uninitialized Data (RAM)|    3FFEA1E0|    3FFF3FA8|   40392
      text|            Cached Code (IRAM)|    40100000|    40105B9D|   23453
irom0_text|           Uncached Code (SPI)|    40209000|    4023ECCB|  220363 <<<
```

so the fs is not overlapping with the image. will look further now

ADiea, 2015-07-13:
Ok, problem solved for me, I used spiffs.exe suggested by @kireevco 

BUT, issue now is that it generates a 192KB spiff.rom isn't this a little too big? can this be configured except recompiling the binary?

kireevco, 2015-07-13:
@ADiea if you you want to configure it - you will have to change Makefile in your sming/Sming directory to `spiffy <whatever_size_in_bytes> <directory>`

kennpie, 2015-07-13:
I'm using a 2014 ESP-01, so 512k. I will try to retest shortly.

robotiko, 2015-07-16:
@kennpie  looks like it is related to #180 
if you go over the spiff limit, it seems that it gets corrupted.. gives no stopping error building/ flashing.. 
but the no files are present.

kireevco, 2015-07-27:
@kennpie hey, any update? :)

kennpie, 2015-08-01:
@kireevco  
Hi Dmitry.  Unfortunately I've been busy at work, so very little fun time. ;-)  I hope to try all recent updates and a newer SDK shortly.

kennpie, 2015-09-02:
sorry for not retrying sooner... I was busy with other stuff.

Anyway - testing today... and working. Thanks.

closed 2015-09-02 by kennpie