SmingHub/Sming #133: HTTPServer_WebSockets example not building

smitjj, 2015-06-23:
Cannot build Websockets example, getting following error.

LD out/build/app.out
out/build/app_app.a(application.o):(.text._Z14startWebServerv+0x38): undefined reference to HttpServer::setWebSocketBinaryHandler(Delegate<void (WebSocket&, unsigned char*, unsigned int)>)' out/build/app_app.a(application.o): In functionDelegate':
/opt/Sming/Sming/SmingCore/Delegate.h:96: undefined reference to `HttpServer::setWebSocketBinaryHandler(Delegate<void (WebSocket&, unsigned char*, unsigned int)>)'
collect2: error: ld returned 1 exit status
make: * [out/build/app.out] Error 1

Any ideas?

robotiko, 2015-06-23:
Same here with latest code from today .
Tested yesterday without issues.

anakod, 2015-06-23:
Pease rebuild SmingFramework project.

robotiko, 2015-06-24:
@anakod, 
I did but didn't solve the issue.

anakod, 2015-06-24:
Try full SmingProject rebuild again please.

robotiko, 2015-06-24:
Fixed!
Thanks a lot

closed 2015-06-24 by anakod