SmingHub/Sming #134: Easy way to get debug information about sming/sdk/environment

kireevco, 2015-06-23:
We need some sort of debug tool that will show what is the version if sming library, what is the sdk and what are other environment values.

@anakod @alonewolfx2 is there anything available from the sdk that will show us right version?
is there any place in sming where we can track version / commit name?
maybe somefile like `_ver.txt` or something?

hreintke, 2015-06-24:
@kireevco : There is a system_get_sdk_version() which shows "1.1.1" in my current config.
Sming version is not (yet) avallable as far as I know

kireevco, 2015-06-24:
@hreintke this is good. is there any way to know it without flashing? (Say we can't flash for some reason and troubleshooting)?

hreintke, 2015-06-24:
@kireevco : Don't unrstand what you are asking. If you didn't flash there is no program and no sdk where that is based on. If you already flashed, it's up to the program to have options to request and display the sdk version where it;s based on.

kireevco, 2015-06-24:
@hreintke I want to know which version of SDK am I currenly pointing via ESP_HOME. This is needed for debug/troubleshooting. 

I'm wondering if sdk version is stored in some binary file somewhere in the sdk, so we could parse it.

hreintke, 2015-06-24:
Ok, now I understand. Will search and try some of my ideas

hreintke, 2015-06-26:
@kireevco : It's getting tricky and not completely clear yet but..
if you take libmain.a and run : ar -x libmain.a
then your (among more a user_interface.o
run (on windows) : \Espressif\xtensa-lx106-elf\bin\xtensa-lx106-elf-objdump.exe -z -D user_interface.o > d.txt -> This is a disassemble(-D) of all functions plus extra (-z) info in d.txt
Look for the : 
00002a60 <system_get_sdk_version>:
    2a60:	ffff21        	l32r	a2, 2a5c <system_uart_swap+0x7c>
    2a63:	f00d      	ret.n
In the lines just above : 
    2a5b:	001c00        	excw
    2a5e:	210000        	srai	a0, a0, 0
00002a60 <system_get_sdk_version>:
See the relation in the 2a5c in the first, thes is/might be the sdk version decoded.
I'm looking further,but do you think that you can use (something like) this for your sdk version check ?

hreintke, 2015-06-26:
Sorry Copy paste error : 
Look for the lines like : 
00002a60 <system_get_sdk_version>:
    2a60:	ffff21        	l32r	a2, 2a5c <system_uart_swap+0x7c>
    2a63:	f00d      	ret.n

hreintke, 2015-06-26:
Sorry, Don't know what causes the wrong display
it should be like 00002a60 system_get_sdk_version

kireevco, 2015-06-28:
it doesn't seem like I have user_interface.o in libmain.a. I'll keep looking.
[image]

hreintke, 2015-06-28:
I am on windows, using SDK206
This is what I gt from libmain.a
```
 Directory of C:\Users\Herman\libs\sdk111

28-06-2015  23:33    <DIR>          .
28-06-2015  23:33    <DIR>          ..
08-06-2015  23:30           137.848 libmain.a
               1 File(s)        137.848 bytes
               2 Dir(s)  140.466.094.080 bytes free

C:\Users\Herman\libs\sdk111>ar -x libmain.a

C:\Users\Herman\libs\sdk111>dir
 Volume in drive C has no label.
 Volume Serial Number is 4AFA-EF19

 Directory of C:\Users\Herman\libs\sdk111

28-06-2015  23:33    <DIR>          .
28-06-2015  23:33    <DIR>          ..
28-06-2015  23:33            18.132 app_main.o
28-06-2015  23:33             1.280 debug-vector.o
28-06-2015  23:33             1.280 double-vector.o
28-06-2015  23:33             9.004 eagle_lib.o
28-06-2015  23:33             6.204 eagle_lwip_if.o
28-06-2015  23:33             6.968 ets_timer.o
28-06-2015  23:33             1.280 kernel-vector.o
08-06-2015  23:30           137.848 libmain.a
28-06-2015  23:33             5.264 mem_manager.o
28-06-2015  23:33             2.132 nmi-vector.o
28-06-2015  23:33             3.908 spi_flash.o
28-06-2015  23:33             3.104 spi_overlap.o
28-06-2015  23:33             1.808 user-vector.o
28-06-2015  23:33            70.588 user_interface.o
28-06-2015  23:33             1.556 vector.o
              15 File(s)        270.356 bytes
               2 Dir(s)  140.465.926.144 bytes free

C:\Users\Herman\libs\sdk111>
```

kireevco, 2015-06-28:
I'm on MacOS and it could be related to the way sdk is build. I'm using esp-open-sdk.

wal-stefan, 2015-08-22:
Looks like Espressif heard your request.
In version 1.2.0 there is a dedicated section in libmain.a for this.
See #207 as well.

`readelf -x .sdk.version  %ESP_HOME%/ESP8266_SDK/lib/libmain.a`

prints (beside others):
```
File: c:/Espressif/ESP8266_SDK/lib/libmain.a(app_main.o)

Hex dump of section '.sdk.version':
  0x00000000 312e322e 3000                       1.2.0. 
```

kireevco, 2015-08-22:
This is beautiful.  @wal-stefan, thank you.

hreintke, 2015-10-03:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-03 by hreintke