SmingHub/Sming #136: Sming Stability & Community uidelines

kireevco, 2015-06-24:
### Stability
I would like to freeze current feature set (pretty rich already), in order to improve stability of existing features and make them bullet-proof. With this Sming will become a production-ready platform that will be rock-solid.

### Documentation
Adding high-level API documentation is another point where we need to improve during this feature freeze. Documentation should not just expose all features but also assure users that all of them are well tested and production-ready.

### Ecosystem
Sming ecosystem is complex by nature. Complexity is added by supporting of non-posix systems. Our goal here is to make everything as much simple for the end-user as possible. One of the ways for doing so is exploring unified environments like docker, where every user will be able to get a working environment quickly and without any trouble. That said, lots of testing and improvement will be required here.

### New contributions
We should still be able to accept contributions into a dev branch but the focus should be on making current feature set in the master branch bullet-proof.

### Community
We will be able to make it together. We will need _YOUR_ help in testing, finding bugs and helping fixing them. Let's make a release and work on testing and bug-fixes only. I hear a lot of people want a stable framework rather than feature-rich buggy one. I assure all of you, a proven record in stability will bring much more new community users. All they want is simply enjoy using sming in their day-to-day hobby or professional life.

### How can YOU help?
- If you think core code needs to be improved - don't wait, improve it and send a PR
- If you have a project built with Sming - share it with us. We'd love to see your code (no matter how good it is)
- If you are facing any issue - [ask community](https://gitter.im/alonewolfx2/Sming), we will try to help! (just make sure to be specific)
- If you agree or disagree with any ticket feel free to express your opinion or at least put a `+`
- If you think you found a bug - [file it](https://github.com/anakod/Sming/issues)! Please make sure information is useful for developers and everything is reproducible ([How to write bug reports](http://www.chiark.greenend.org.uk/~sgtatham/bugs.html))
- If you think your experience is not smooth enough (not enough clarity, instructions are or not self explanatory) - it should be [filed](https://github.com/anakod/Sming/issues) as well

#### _Remember, if a user is having hard time - it's a bug!_

Sming is a framework that is created by community for community.
We value all of YOU, creative, innovative, smart and just simply great people here at _anakod/sming_

_Feel free to start a discussion here_

alon24, 2015-06-24:
+

ystrem, 2015-06-25:
Very nice. :)

Bravo13, 2015-06-25:
+

robotiko, 2015-06-25:
Nice +

Tavalin, 2015-06-25:
I agree @kireevco.  
A lot of effort has gone in already, but there's still a lot to do and the community/userbase will be key to getting it to the level we want through their feedback and contributions.

prussiap, 2015-06-25:
+10 also. Fully agreed and very appreciative to @kireevco for taking the lead. He's super busy with work and life so I hope others will join in and help stabilize this with PRs and bug fixes..

Most of the below is known but I'm reiterating from a software dev point of view:
I would say give this a branch for stable release.. a whole number or half number (1 or 1.5) for the final version you're working on.. once that's out you can do features off that main stable merged branch.

If anybody branches off into some new fancy features make sure to make only one branch for dev or feature requests and to only commit it once it's tested and somewhat stable...

Nothing worse then a lot of orphan branches .. Thankfully Sming has been pretty good at avoiding this stuff.

Otherwise I think the guidelines and goals are fantastic and I hope I can find some time to contribute a bit.

kennpie, 2015-06-27:
+1

I'm not the strongest programmer in C or C++, but right away I found that it was far easier to get working results with Sming, over struggling with just the Espressif C SDK, and now that I've been using it and trying to stay up-to-date with your development, I remain happy with my choice to use it. 

If time permits, I will try to test more, submit bugs, and help out with documentation.

Thanks again for a great project.

kireevco, 2015-08-23:
Git flow proposal: #230 related.

hreintke, 2016-01-27:
Solved by release process and documentation activity

closed 2016-01-27 by hreintke