SmingHub/Sming #137: httpclient not working correctly on shared hosting with parameter.

alonewolfx2, 2015-06-24:
for examle here is working request
```
String request="http://www.mbozkurt.com/pgc.php";
httpclienttt.downloadString(request, onDataSent);
```
result is "adam icerde"
but here is not working request.
```
String request="http://www.mbozkurt.com/pgc.php?tagid=test";
httpclienttt.downloadString(request, onDataSent);
```
result is 
```
Failed
Server response: '<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>404 Not Found</title>
</head><body>
<h1>Not Found</h1>
<p>The requested URL /pgc.php was not found on this server.</p>
<p>Additionally, a 404 Not Found
error was encountered while trying to use an ErrorDocument to handle the request.</p>
</body></html>
'
```
it seems host header missing or parse error. i tried to change uri.host in httpclient.c and same result. how can i solve this problem?

kireevco, 2015-06-24:
I'm using telnet for my test. I sent `GET http://www.mbozkurt.com/pgc.php?tagid=deneme\r\nHTTP/1.1` and got a valid response back (`adam icerde`).

alonewolfx2, 2015-06-24:
@kireevco soo problem about httpclient not hosting provider right?

kireevco, 2015-06-24:
@alonewolfx2 seems like that. I don't know any more minimalistic test than sending a pure TCP request.

We need to test http client on a local server with full logging enabled. To see what's coming in.
Something like `ruby -r sinatra -e "get('/') { puts params.inspect }"`

anakod, 2015-06-25:
We need full client request dump to solve this issue.

alonewolfx2, 2015-06-25:
@anakod ho can i make full client request dump? or can you make ? it is simple to re-try

kireevco, 2015-06-25:
@alonewolfx2 you need some sort of local server, where you can control verbose logging. like sinatra or flask or maybe even IIs if you can make it work

anakod, 2015-06-25:
Please try http://echo.httpkit.com it will dump your request

alonewolfx2, 2015-06-25:
ok here we go.
```
tagId-deneme
req:http://echo.httpkit.com/pgc.php?tagid=denemedeneme

Failed
Server response: '<html>
<head><title>404 Not Found</title></head>
<body bgcolor="white">
<center><h1>404 Not Found</h1></center>
<hr><center>nginx/1.1.19</center>
</body>
</html>
'

tagId-deneme
req:http://echo.httpkit.com
Success sent
Server response: '{
  "method": "GET",
  "uri": "/",
  "path": {
    "name": "/",
    "query": "",
    "params": {}
  },
  "headers": {
    "x-forwarded-for": "78.xx.72.xxx",
    "host": "echo.httpkit.com"
  },
  "body": null,
  "ip": "127.0.0.1",
  "powered-by": "http://httpkit.com",
  "docs": "http://httpkit.com/echo"
}'
```

liquiddandruff, 2015-06-26:
@kireevco the get request you did was not using relative paths, you've specified the host indirectly with the full url.

alonewolfx2, 2015-06-26:
@anakod @kireevco @liquiddandruff  nodemcu working fine with same url same parameter.

anakod, 2015-06-26:
> req: without parameter but with host header
How did you specify it?

liquiddandruff, 2015-06-26:
Made a new line and set Host: mbozkurt.com. It's just HTTP protocol really.

anakod, 2015-06-26:
I'm not understand, I mean how it was setted in this test (in code, I think)

liquiddandruff, 2015-06-26:
Oh yeah, not sure about that. Not sure what that dump from httpkit means soo :laughing:

anakod, 2015-07-01:
Very strange. I have checked it and it working:
`thingSpeak.downloadString("http://echo.httpkit.com/pgc.php?tagid=denemedeneme", onDataSent);`
(Only one line changed in standard ThingSpeak example)

```
Success sent
Server response: '{
  "method": "GET",
  "uri": "/pgc.php?tagid=denemedeneme",
  "path": {
    "name": "/pgc.php",
    "query": "?tagid=denemedeneme",
    "params": {
      "tagid": "denemedeneme"
    }
  },
  "headers": {
    "x-forwarded-for": "82.193.155.240",
    "host": "echo.httpkit.com"
  },
  "body": null,
  "ip": "127.0.0.1",
  "powered-by": "http://httpkit.com",
  "docs": "http://httpkit.com/echo"
}'
Sensor value wasn't accepted. May be we need to wait a little?
```

`thingSpeak.downloadString("http://www.mbozkurt.com/pgc.php?tagid=test", onDataSent);`
Response:
```
Success sent
Server response: 'adam icerde'
Sensor value wasn't accepted. May be we need to wait a little?
```

alonewolfx2, 2015-07-09:
it seems some shared hostings very selective. i am closing this issue because it seems not directly related sming

closed 2015-07-09 by alonewolfx2