SmingHub/Sming #140: Wdt Reset on closing and reopening tcp server connection

M0ebiu5, 2015-06-26:
How to reproduce:

- take any example with a tcp server or http server
- after tcpserver.listen(80) add: tcpserver.close(); tcpserver.listen(80);
- flash and crash

hreintke, 2015-06-26:
@M0ebiu5 : What version of sming/sdk are you using 
Mine are : Windows7,UDK 206,  latest sming (25/6/15),sdk 1.1.1
Do you always have WDT reset ?
In my test, the close() en listen(port) doesn't give errors but after trying to use (connect) to the restarted connection I get NO CONNECTION ON TCP.
Not correct either but different symptoms..

M0ebiu5, 2015-06-26:
Good question - i couldnt find some version info in the sming files. 
It's apr. 4 weeks old. Sdk: v1.0.1_15_04_24. Win 8.1 x64

I've tried with the latest sming version now (26/6/15) and i get the same behavior like you.

Loooopy, 2015-07-13:
Check out this: http://tech.scargill.net/esp8266-debug/
Since Espressif SDK 1.01 there are some api changes.

New watch dog functions:
void pp_soft_wdt_stop(); // close software watchdog
void pp_soft_wdt_restart(); // reset software watchdog

I did a try my self but did not get it to work, 
so i do not know if there was something i did or the things are broken.

hreintke, 2015-11-29:
SDK 1.1.1 related. If still exist on Sming 1.4.0 or Sming develop please submit a new issue.

closed 2015-11-29 by hreintke