SmingHub/Sming #142: Wdt Reset when running udp, tcpserver and udpclient at once

M0ebiu5, 2015-06-27:
When running udp and tcpserver and udpclient at once, the app becomes unstable after apr. 70 minutes. An http request after that time causes a wdt reset.
Stressing the udp or tcp connection does not significantly change the timing of the crash. Heap is never lower than 23kb.
Only without any request it survives.

Sming ver: 26.6.15 - sdk v1.0.1_15_04_24

How to reproduce:
- flash application: https://github.com/M0ebiu5/EspVent  (adapt router and password in init())
- make apr. 15 http requests
- wait for more than apr. 70 minutes
- make http request
- wdt reset

Tavalin, 2015-06-27:
Are you by any chance using SystemClock?

M0ebiu5, 2015-06-27:
Thanks for the suggestion - i will try it out and report back.
My tests showed, that if i dont use the .listen() method on udpconnection, the app is stable.

Tavalin, 2015-06-27:
You mentioned it's roughly 70mins, the SystemClock bug occurs at 1hr 11mins, 34seconds as that's when millis() overflows and throws SystemClock.now() into an infinite loop.  Just spent the last week looking at / fixing it and I'm 99% sure it's the same issue.  

If you comment out calls to SystemClock.getSystemTimeString() for now you should be OK.

M0ebiu5, 2015-06-28:
I've retested it and it is the SystemClock bug.
Thanks!

closed 2015-06-28 by M0ebiu5