SmingHub/Sming #143: Add access to httprequest body

alon24, 2015-06-28:
I want to use json with web, and send data from webpage (supplied by esp) to the server (esp server), using ajax.

I want to use AJAX to trigger a command on the server and use JSON to tell its params and data.
There is no way on the server side to reach the body of the request and parse the JSON payload.

For instance, have a button to toggle a relay (from web) and return the current state in the response.

kireevco, 2015-06-28:
@alon24 What stops you from doing it right now?

alon24, 2015-06-29:
Hi,

The fact that I do not know how, and I asked in gitter and they said it
needs to be added, so like a good sport I opened a bug request (did not
know how to set it as feature request).

alon24, 2015-07-13:
is there a way to access the body?

dereulenspiegel, 2015-09-07:
I have the exact same problem. Parsing the body as POST request parameters is not always useful. What would be the preferred way of solving this in Sming? We could for example not parse the body by default, but supply method on the request object which parses the body if needed. Or we keep a copy of the body around. But that doesn't sound very efficient.

dereulenspiegel, 2015-09-08:
Just for information: I implemented a first attemp to solve this problem (see here https://github.com/dereulenspiegel/Sming/tree/http-body). I will create a pull request from this, since it shouldn't break anything.

hreintke, 2015-12-21:
Working on a solution in PR #487.
Please continue discussion there

closed 2015-12-21 by hreintke