SmingHub/Sming #144: How does the buffer in HardwareSerial work?

0xPIT, 2015-06-28:
I can not find where pRcvMsgBuff is ever allocated. Yet it seems to be written to from the ISR.
I'm currently refactoring the code for use with the RTOS SDK. 
Can you please explain how it is intended to work, for sending and receiving?

hreintke, 2015-06-28:
@0xPIT : I think is is defined & allocated in hardwareserial.cpp
line 17/18 : 
```
// UartDev is defined and initialized in ROM code.
extern UartDevice UartDev;
```
and used on line 39
```
	ETS_UART_INTR_ATTACH((void*)uart0_rx_intr_handler,  &(UartDev.rcv_buff));
```

0xPIT, 2015-06-28:
Yes, I've seen that. This still does not allocate any memory to the member `pRcvMsgBuff`of the struct `rcv_buff`, that is a member of `UartDevice`

What does that mean `UartDev is defined and initialized in ROM code`?
`grep` does not reveal anything.

hreintke, 2015-06-28:
I assume (yes, I know it is dangerous and a reason to get errors), it's part of esp8266 startup code.
Which seems reasonable as the uart is initialized in very early stages of system startup as it outputs state/progress information then.

AlmightyFrog, 2015-06-28:
Some month ago i asked myself same question - also for me it looked like it's initialised by esp8266 startup. We do not have controll over that - that's the price paying for 3rd party blobb stuff.

But what do you see dangerous there @hreintke? Sure, we have no controll over that, but it's a buffer which is already there and as it's not the hardest stuff to code i think it should be kind of bug free.

Of course we could also implement own buffer but i do not see benefit there.

hreintke, 2015-06-28:
@AlmightyFrog : Sorry to confuse you. I mend dangerous by doing assumptions in general, not specific to this hardware serial question

0xPIT, 2015-06-29:
Ah, got it. It's not present in the RTOS-SDK. In the regular SDK (1.1.2) it's provided in the linker file:
`ld/eagle.rom.addr.v6.ld:PROVIDE ( UartDev = 0x3fffde10 );`

hreintke, 2015-10-03:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-03 by hreintke