SmingHub/Sming #152: Community projects with Sming

kireevco, 2015-06-30:
We all love Sming, and I'd like us to start sharing what we have done with it so far.
Feel free to post a link to your github repo with your project.
Please make sure to have at least a README.md in your repo. Pictures, schematics and videos are WELCOME!

alon24, 2015-06-30:
This is the begining of an esp8266 SousVide controller :
https://github.com/alon24/esp8266_projects

colinmeikle, 2015-07-07:
Here is an IR remote library (receive only)
https://github.com/colinmeikle/nec_ir_rec

it uses the the nec protocol, which is common on some of the cheep ebay remotes.

hreintke, 2015-07-07:
@colinmeikle : I just look ed at your code as I was curious how you implemented and noticed the delegate/callback distinction you made in 
```
/**
 * sets the callback function which should tack a char parameter i.e. the key
 * @param callback
 */
void IR_rec::setIRCallBack(IRCallback callback){
    recCallback=callback;
       recDelegateCallback=null;
}
/**
 * sets the Delegate callback function which should tack a char parameter i.e. the key
 * @param callback
 */
void IR_rec::setIRCallBack(IRRecDelegate callback){
    recDelegateCallback=callback;
    recCallback=null;
}
```
If you only define the IRRecDelegate one, You still should be able to call it with the callback function.
The definition using the typedef Delegate<void(char c)> IRRecDelegate;
is able to hold both "global functions" as "class member functions"

colinmeikle, 2015-07-07:
Thanks hreintke, I had never used Delegates before so thanks for your comments. I had seen another example like this so i thought I needed separate functions. I'll try it out just with the IRRecDelegate and see if it works. Thanks for comments anything that helps me improve my code is good!

colinmeikle, 2015-07-27:
managed to get my heart rate sensor working with sming without any trouble  :-) https://github.com/colinmeikle/heartrate/tree/master . This is for the sensor found here http://pulsesensor.com/ and is simply a mod of the ardiuno library for this sensor

hreintke, 2015-10-03:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-03 by hreintke