SmingHub/Sming #154: TCP / HTTP doesn't appear to be stable

kireevco, 2015-07-03:
In HTTP_Config example device is easily crashable. 
Can we add more protection/make code more robust?

anakod, 2015-07-03:
Please, provide more details.

kireevco, 2015-07-06:
Here is my test case: http://screencast.com/t/cDmzFVijMry (video)
Here is my flash/terminal log:
```
Killing Terminal to free /dev/ttyUSB0
make: *** [flash] Killed
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 921600 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin 0x44000 out/firmware/spiff_rom.bin
Connecting...
Erasing flash...
Writing at 0x00008000... (100 %)
Erasing flash...
Writing at 0x00041c00... (100 %)
Erasing flash...
Writing at 0x00073c00... (100 %)

Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200
--- Miniterm on /dev/ttyUSB0: 115200,8,N,1 ---
--- Quit: Ctrl+]  |  Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
 23696, room 16 
;l+TCP connection
timeout updating: 70 -> 65535
Server timeout updating: 40 -> 90
+TCP connection
timeout updating: 70 -> 65535
Server timeout updating: 40 -> 900
fs.start: size:224 Kb, offset:0x44000

Set AP configuration in background
mode : sta(18:fe:34:a4:57:c9) + softAP(1a:fe:34:a4:57:c9)
add if0
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100
addUser: me 123
'/' registered
'/ipconfig' registered
'/ajax/get-networks' registered
'/ajax/connect' registered
bcn 0
del if1
add if1
bcn 100
AP configuration was updated on system ready event
f 0, scandone
scan completed: 33 found
add 1
aid 1
station: b8:e8:56:38:64:22 join, AID = 1
onAccept state: 0 K=0
Free heap size=24440, K=0
+TCP connection
timeout updating: 70 -> 90
Tcp Server onClient 
path=/
Host === 192.168.4.1
parsed
Request: GET, nodata
TCP received: 338 bytes
onReadyToSendData: 1
send file: index.html (12168 bytes)
response sendHeader
response sendBody
READ Template (0)
plain template text pos: 0, len: 1024
READ Template (0)
plain template text pos: 1024, len: 1024
READ Template (0)
plain template text pos: 2048, len: 809
TCP sent: 1460
onReadyToSendData: 2
[... 218 lines cut ...]
realloc 3119 -> 3185
realloc 3184 -> 3250
realloc 3249 -> 3315
realloc 3314 -> 3380
realloc 3379 -> 3445
realloc 3444 -> 3510
realloc 3509 -> 3575
realloc 3574 -> 3640
realloc 3639 -> 3705
realloc 3704 -> 3770
realloc 3769 -> 3835
realloc 3834 -> 3900
realloc 3899 -> 3965
realloc 3964 -> 4030
No heap available, failed to malloc 4048
Fatal exception (29):
epc1=0x4023a21c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000f7d, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```

hreintke, 2015-07-08:
@kireevco : 
I checked the MemoryDataStream::write function. There is the memory buffer increased if capacity isn't enough for adding new data. There is however no check on memory fail on that increase.
Below is an adapted vrersion of the function. If no more memory available It just truncates the message, so probably still "application errors" but the crash should not happen.
Could you update the function in DataSourceStream.cpp and redo your test ?
Depending on the outcome we need to see what updates are further necessary.
```
size_t MemoryDataStream::write(const uint8_t* data, size_t len)
{
	//TODO: add queued buffers without full copy
	if (buf == NULL)
	{
		buf = (char*)malloc(len + 1);
		buf[len] = '\0';
		memcpy(buf, data, len);
	}
	else
	{
		int cur = size;
		int required = cur + len + 1;
		if (required > capacity)
		{
			capacity = required < 256 ? required + 128 : required + 64;
			debugf("realloc %d -> %d", size, capacity);
			char* reallocBuf = (char*)realloc(buf, capacity);
			if (!reallocBuf)
			{
				debugf("Memory fail on realloc");
				// buf is still pointing to valid memory
				len = capacity - size - 1; // fill up to end of capacity
			}
			else
			{
				buf = reallocBuf;
				// buf is invalidated by realloc, reallocBuf is pointing to new area
			}
		}
		buf[cur + len] = '\0';
		memcpy(buf + cur, data, len);
	}
	pos = buf;
	size += len;
	return len;
}
```

koltegirish, 2015-09-04:
hello folks,
I seen this kind of output on terminal, when using both sniffer mode and MQTT .Any idea how fix this issue.

Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Management Packet (len=128) Sni
                               enable promiscuous mode
RSSI: -94
Starting of loop
Data Packet (len=60) Sniffed on Channel 11
cnt: 1
rdaddr: 01:00:5e:00:00:fc tsaddr: b0:c5:54:82:c5:1a bssid: f8:2f:a8:ce:8d:9b addr4: 35:35:c9:60:01:00
Entering in sleepTimer function
Step 2
Step 3
Step 4
Connecting to MQTT Server...

load ...
MQTT_InitConnection
MQTT_InitClient
WIFI_INIT
enable promiscuous mode
RSSI: -94
Starting of loop
Data Packet (len=60) Sniffed on Channel 11
cnt: 1
rdaddr: 01:00:5e:00:00:fc tsaddr: b0:c5:54:82:c5:1a bssid: f8:2f:a8:ce:8d:9b addr4: 35:35:ca:60:01:00
Entering in sleepTimer function
Step 2
Step 3
Step 4
Connecting to MQTT Server...

load ...
MQTT_InitConnection
MQTT_InitClient
No heap available, failed to malloc 2064
WIFI_INIT
No heap available, failed to malloc 904
No heap available, failed to malloc 904
enable promiscuous mode
RSSI: -96
Starting of loop
Management Packet (len=128) Sniffed on Channel 11
Starting of loop
Data Packet (len=60) Sniffed on Channel 11
cnt: 1
rdaddr: 33:33:00:01:00:03 tsaddr: b0:c5:54:82:c5:1a bssid: f8:2f:a8:ce:8d:9b addr4: 35:35:cb:60:01:00
Entering in sleepTimer function
Step 2
Step 3
Step 4
Connecting to MQTT Server...

load ...
MQTT_InitConnection
MQTT_InitClient
No heap available, failed to malloc 1040
No heap available, failed to malloc 1040
No heap available, failed to malloc 2064
WIFI_INIT
No heap available, failed to malloc 904
No heap available, failed to malloc 904
e

hreintke, 2016-01-27:
Closed due to long time no activity

closed 2016-01-27 by hreintke