SmingHub/Sming #155: Sming Manual Windows Installation page isn't correct

lawrence-jeff, 2015-07-03:
Maybe I am missing something but this page doesn't seem right
https://github.com/anakod/Sming/wiki/Windows---Manual-Installation
Step 2 has you use a MinGW path c:\tools - but if you follow the instructions before that you don't have a C:\tools, the default installs it to C:\mingw
Also the example needs quotes around the path or it fails with a setx error
Step 3 has you set a variable referring to C:\tools\Sming but no step in the process before (or after) actually downloads SMING and puts it in the c:\tools directory. 

Also if you follow the chocalate thing and try to install just the sming examples (per the cmd). It doesn't work as is because its missing the closing quote and for some reason the examples command tries to install all the prerequisites

kireevco, 2015-07-03:
@lawrence-jeff Preferred way is using chocolatey, and it changes quite often under the hood.
Manual installation is provided for reference and assumes you know how mingw/eclipse environment works and specifically directory paths can differ across systems.

I updated [Windows-Quickstart](https://github.com/anakod/Sming/wiki/Windows-Quickstart), so everyone can actually see what's happening under choco sming package.

Please consider retrying choco, make sure to use --force:
```
choco install sming -y -source 'https://www.myget.org/F/kireevco-chocolatey/' -force
```

Also, I would recommend trying [sming-docker](https://github.com/kireevco/sming-docker) as it elimitates a lot of enviromental issues.

lawrence-jeff, 2015-07-04:
I will take a look at the chocoa install - I already had most things installed so I was only looking at how to add Sming alone which is why I ended up on the manual page (the choco piece tries to reinstall 64 bit pieces of everything when I have the 32 bit installed and working per the UDK instructions)

Not to belabor a point and nitpick because I'm sure your time is better spent on code, but if you follow the instructions in step 1 - the path doesn't 'differ' - it will always be c:\mingw (since that is the default in the install the instructions say to run and they specify some options but not a path) not c:\tools\mingw
How about editing that line to say:
Install MingGW. Run mingw-get-setup.exe. Add install support for the graphical user interface. Change the install path to c:\tools\mingw

That way if you are following the instructions as is everything lines up

Also it seems in a page about manually installing Sming really needs to include a step on how to manually install Sming. The page has all the prerequisites but never tells you how to install it, it gets to step 3 as if you downloaded or installed something but you haven't if you are following along, do you just download the whole archive and extract to c:\tools\sming? The reference to a c:\tools\sming\sming makes it a bit more confusing.

Anyway appreciate all the great work just trying to point out some things that can be confusing on the wiki

kireevco, 2015-07-04:
Appreciate your feedback, will adjust the path in the wiki!

Installing sming is just getting the repo (via git or zip)

Also setting environment variables is very important.

You will also need msys-make.

kireevco, 2015-07-04:
Please check the most recent version and let me know what you think https://github.com/anakod/Sming/wiki/Windows---Manual-Installation

kireevco, 2015-07-11:
I'm closing this ticket, seems that everyone is good.

closed 2015-07-11 by kireevco