SmingHub/Sming #156: Migrate sming repo under opensource group

kireevco, 2015-07-03:
I'm proposing to move sming repo to https://github.com/esp-sming/sming
This way we will be able to add multiple integrations easily (travisci, pivotal, etc).
It will also related to https://github.com/anakod/Sming/issues/69 and https://github.com/anakod/Sming/issues/136

@anakod will need to transfer repo as is with all issues and wiki.

Let me know what you think folks.

ystrem, 2015-07-05:
Yep +1

Loooopy, 2015-07-13:
+1
Sounds good, it will be easier to find for new guys.

zhivko, 2015-07-13:
I vote for it. +1

kennpie, 2015-07-13:
As long as n00bs like me can still get it, sure.

kireevco, 2015-07-27:
@anakod any update on this?

robotiko, 2015-09-21:
Any update on this?

kireevco, 2015-09-28:
Done! Thank you @anakod 
(Hi from Köln)
[image]

closed 2015-09-28 by kireevco