SmingHub/Sming #161: debugf formatter doesn't handle floats

robotiko, 2015-07-06:
when trying to output float values  with float formatter such as %.2f %3.2f  %f  if simply outputs the formatter without parsing it neither the values.

robotiko, 2015-07-06:
Tested, same issue with Serial.printf()
Seems to be a limitation of the printf (and related sn/vsn) functions provided by espressif.

ADiea, 2015-07-06:
@robotiko Indeed, use Serial.print(float_value) instead

robotiko, 2015-07-08:
@ADiea it is ok for temporary workaround, however debugf should not be hardcoded to Serial because as stream could be redirected somewhere else that is not the serial port (there is a work in progress about more advanced debug and this will conflict there).

hreintke, 2015-07-09:
@robotiko : I worked today with some strings and found another way of debugf usage:
```
	float t = 3.3;
	debugf("Float value =  %s",String(t).c_str());
```

robotiko, 2015-07-09:
@hreintke that works ok and respects the stream flow. :dancer:

anakod, 2015-07-23:
It is standard embedded printf limitation (because floating point printf implementation is too big).
So I think it isn't bug.

ADiea, 2015-07-24:
i am working on a version that uses sming/sming/system/stringconversion->dtof() until espressif updates os_printf

hreintke, 2015-07-24:
@ADiea : I am working on a debug implementation which is also capable to output to other then serial (f.e telnet, websocket). this makes use of the Wirinf/Print.h, print.cpp functionality in sming.
The wring implementation has the same limitation on the printf as debugf/os_printf because it is based on the ets_vsnprintf from espressif. If you can implement your solution in the Wiring printf is has the advantage of : 1/ Not only debugf will have the functionality but also f.e Serial.printf and 2/ The new debug system will automatic have the same functionality.

ADiea, 2015-07-24:
@hreintke I will try and see how this impacts the codesize

in my project I commented out all Serial references and avoid using the wiring lib, this saved me several K of codespace and ~1K of RAM if I remember correctly. 

IMHO as long as the libs from espressif already depend on ets_vnprintf etc so these already link in the final binary, I don't see the use of the wiring library/HardwareSerial classes, as they only duplicate code...unfortunately I see the json library also uses wiring :(

I like these classes as they offer building blocks to build upon but on this tiny device they seem overkill to me, I'm trying to use what's already provided by espressif in their libs...

I will give it a go though and see the difference, will also make a PR soon so we can discuss more there

ADiea, 2015-07-24:
@hreintke i opened #188 also included the wiring Print class modification to use the custom vsprinf need to test it more and maybe adjust the basic console example or smth so still wip. It is open to discussion

hreintke, 2015-10-05:
Sming repository is currently being restructured and cleaned. This issue has been inactive for a long period and is being closed. If the issue is still valid please open a new with complete description

closed 2015-10-05 by hreintke