SmingHub/Sming #162: Makefile issue following Windows Quickstart

Bennedetto, 2015-07-06:
I am using Windows 7 Pro. I installed Sming following your windows quickstart guide using Chocolatey (By the way, there is one inverted comma missing at the end of the examples installation).
Starting the make, i get the following error:

C:\tools\sming\Sming/Makefile-project.mk:49: /C/tools/sming/Sming/Makefile-windows.mk: No such file or directory
make: *** No rule to make target `/C/tools/sming/Sming/Makefile-windows.mk'.  Stop.

Cheers
Benne

Def152, 2015-07-06:
Hi, you are not alone :P I have the same problem.

kireevco, 2015-07-06:
@Bennedetto 
1. Please try executing `make` from cmd from your project's directory
2. Please open cmd.exe, execute following commands and report back:
```
echo %SMING_HOME%
echo %ESP_HOME%
echo %PATH%
dir %SMING_HOME%
dir %ESP_HOME%
```

@Bennedetto Also, please consider using [sming-docker](https://github.com/kireevco/sming-docker) as it doesn't have most of unique environment issues.

Bennedetto, 2015-07-06:
this is are the results (fresh like generated from Chocolatey - without any changes):
**1. Please try executing make from cmd from your project's directory**
```
C:\tools\sming.examples\Basic_Blink>make
C:\tools\sming\Sming/Makefile-project.mk:49: /C/tools/sming/Sming/Makefile-windo
ws.mk: No such file or directory
make: *** No rule to make target `/C/tools/sming/Sming/Makefile-windows.mk'.  St
op.
```
**echo %SMING_HOME%**
```
C:\tools\sming.examples\Basic_Blink>echo %SMING_HOME%
C:\tools\sming\Sming
```
**echo %ESP_HOME%**
```
C:\tools\sming.examples\Basic_Blink>echo %ESP_HOME%
c:/Espressif
```
**echo %PATH%**
```
C:\tools\sming.examples\Basic_Blink>echo %PATH%
C:\WinAVR\bin;C:\WinAVR\utils\bin;C:\ProgramData\Oracle\Java\javapath;C:\Windows
\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerSh
ell\v1.0\;C:\Program Files\OpenVPN\bin;C:\Program Files\Microsoft SQL Server\110
\Tools\Binn\;C:\ProgramData\chocolatey\bin;C:\Program Files\Eclipse Foundation\4
.4.2\eclipse;C:\tools\mingw64\bin;C:\tools\mingw64\msys\1.0\bin;C:\tools\mingw64
\bin;
```
**dir %SMING_HOME%**
```
C:\tools\sming.examples\Basic_Blink>dir %SMING_HOME%
 Volume in Laufwerk C: hat keine Bezeichnung.
 Volumeseriennummer: 5868-F352

 Verzeichnis von C:\tools\sming\Sming

06.07.2015  11:11    <DIR>          .
06.07.2015  11:11    <DIR>          ..
25.06.2015  05:53             8.597 .cproject
25.06.2015  05:53               815 .project
06.07.2015  11:11    <DIR>          .settings
06.07.2015  11:11    <DIR>          appinit
06.07.2015  11:11    <DIR>          apptest
06.07.2015  11:11    <DIR>          compiler
06.07.2015  11:11    <DIR>          include
06.07.2015  11:11    <DIR>          Libraries
25.06.2015  05:53             7.687 License.txt
25.06.2015  05:53             8.323 Makefile
25.06.2015  05:53               554 Makefile-bsd.mk
25.06.2015  05:53               541 Makefile-linux.mk
25.06.2015  05:53               533 Makefile-macos.mk
06.07.2015  12:06            13.841 Makefile-project.mk
25.06.2015  05:53               540 Makefile-windows.mk
06.07.2015  11:11    <DIR>          Services
06.07.2015  11:11    <DIR>          SmingCore
06.07.2015  11:11    <DIR>          system
06.07.2015  11:11    <DIR>          Wiring
               9 Datei(en),         41.431 Bytes
              12 Verzeichnis(se), 47.110.434.816 Bytes frei
```
**dir %ESP_HOME%**
```
C:\tools\sming.examples\Basic_Blink>dir %ESP_HOME%
Ungültige Option - "Espressif".
```

Since ESP_HOME seems to be wrong, i corrected that value and here is the new output:
**echo %ESP_HOME%**
```
C:\Users\Toshiba>echo %ESP_HOME%
C:\Espressif
```
**dir %ESP_HOME%**
```
C:\Users\Toshiba>dir %ESP_HOME%
 Volume in Laufwerk C: hat keine Bezeichnung.
 Volumeseriennummer: 5868-F352

 Verzeichnis von C:\Espressif

06.07.2015  11:09    <DIR>          .
06.07.2015  11:09    <DIR>          ..
10.06.2015  17:20            16.130 changelog.txt
21.05.2015  21:52             2.061 Command line Unofficial Development Kit for
Espressif ESP8266.lnk
16.01.2015  09:28               186 DevKit web site.url
06.07.2015  11:09    <DIR>          docs
06.07.2015  11:09    <DIR>          ESP8266_RTOS_SDK
06.07.2015  11:09    <DIR>          ESP8266_SDK
06.07.2015  11:09    <DIR>          ESP8266_SDK_094
06.07.2015  11:09    <DIR>          ESP8266_SDK_095
06.07.2015  11:09    <DIR>          ESP8266_SDK_101
06.07.2015  11:09    <DIR>          ESP8266_SDK_110
21.11.2014  11:22               180 Espressif web site.url
06.07.2015  11:09    <DIR>          examples
06.07.2015  11:09    <DIR>          extra
25.03.2015  09:16           137.750 Install.ico
06.07.2015  11:09    <SYMLINKD>     sdk [c:\Espressif\ESP8266_SDK]
06.07.2015  11:09           647.847 unins000.dat
06.07.2015  11:08         1.312.112 unins000.exe
06.07.2015  11:09            23.001 unins000.msg
21.11.2014  10:45           147.878 Uninstall.ico
06.07.2015  11:09    <DIR>          utils
06.07.2015  11:08    <DIR>          xtensa-lx106-elf
               9 Datei(en),      2.287.145 Bytes
              14 Verzeichnis(se), 47.109.931.008 Bytes frei
```
All the other results stayed (as expected) the same..

kireevco, 2015-07-06:
Please make sure __ESP_HOME__ is set to `c:\Espressif` (normal Windows backslash)
You should also try [force reinstall](https://github.com/anakod/Sming/wiki/Windows-Quickstart#force-reinstall-sming) sming. It should take care of all settings.

Essentially:
```powershell
choco install sming -y -force -source 'https://www.myget.org/F/kireevco-chocolatey/'
```

Bennedetto, 2015-07-06:
I corrected ESP_HOME -> c:\Espressif and forced the reinstallation of sming. but the error stays the same..

kireevco, 2015-07-06:
Please confirm you did all proper environment refreshes and your `echo %ESP_HOME%` is showing new value. Also, confirm you are still testing `make` from the command line.

kireevco, 2015-07-06:
Please do one more troubleshooting step: `where make` and post output here.

kireevco, 2015-07-06:
```where make```

Bennedetto, 2015-07-06:
Yeha, got it. The problem was the already existing installation of WinAVR on my machine. I checked ```where make``` and the result was
```
C:\Users\Toshiba>where make
C:\WinAVR\utils\bin\make.exe
C:\tools\mingw64\msys\1.0\bin\make.exe
``` 
So there were 2 versions existing - and i guess just the wrong one got choosen. So I removed (zip and backup) the whole ```C:\WinAVR\utils\bin``` folder to get rid of that problem. Now it works like a charm =)!
Thanks!!!

kireevco, 2015-07-06:
@Bennedetto I'm glad that you worked it out.
This is a unique environment issue, that's why I recommend using docker - it doesn't have such issues.

I'm closing this issue then!

closed 2015-07-06 by kireevco