SmingHub/Sming #164: Feature Request: Websocket client support

robotiko, 2015-07-06:
There are some very useful usages of a Websocket client that would make it an interesting addition. 

Considering the limitations of number of connections accepted in the Websocket server (5 I guess),  there is no way to create a realtime sensor mesh using websockets bigger than that. 

Acting as client, we can have a host acting as a server and clients register from Local wifi  without complex NAT routes, simply connect each device to the server so we can have something similar to HttpClient example but with realtime two way comm with very simple setup.

Also requested in #145

colinmeikle, 2015-07-07:
I agree websocket client support would be useful . Note however if you can use standard tcp sockets the TcpClient_NarodMon gives an example of doing this using tcp sockets.

kireevco, 2015-07-07:
My opinion is that TCP has to be improved and be more robust before relying on web socket implementation.

hreintke, 2015-07-07:
@kireevco : If we can instruct all who finds issues on tcp networking to add those to the https://github.com/anakod/Sming/issues/154 then are getting at least an overview of them.
I will be out a couple of days fri-wed but then can have a look on those. 
Of course no guarantee that i actually am able to fix.

azmathmoosa, 2016-02-12:
Has anyone done anything?  Are there any alternatives?!

KSDaemon, 2016-03-16:
Voting for this feature. I'm also seeing interesting use cases for this.

hreintke, 2016-04-22:
Duplicate of #418

closed 2016-04-22 by hreintke