SmingHub/Sming #165: Sming Build Error on SmingCore/Platform/RTC

smitjj, 2015-07-06:
When building Sming, the following build error occurs.

```
C+ SmingCore/Platform/RTC.cpp
SmingCore/Platform/RTC.cpp: In constructor 'RtcClass::RtcClass()':
SmingCore/Platform/RTC.cpp:4:39: error: 'system_get_rst_info' was not declared in this scope
  rst_info* info = system_get_rst_info();
                                       ^
SmingCore/Platform/RTC.cpp:5:35: error: 'REASON_WDT_RST' was not declared in this scope
  hardwareReset = (info->reason == REASON_WDT_RST);
                                   ^
make: *** [out/build/SmingCore/Platform//RTC.o] Error 1
```

Tavalin, 2015-07-06:
What version of SDK are you using?

smitjj, 2015-07-07:
sdk_v1.1.0_15_05_26

hreintke, 2015-07-07:
searching SDK's, according changelogs and direcory contents : 
system_get_rst_info and REASON_WDT_RST are only available from SDK 1.1.1 onwards

smitjj, 2015-07-07:
I will update my SDK and check. Sming dependency then need to be updated from Espressif SDK v1.0.1b1 to v1.1.1

smitjj, 2015-07-07:
I can confirm it build successfully with 1.1.2, https://bintray.com/artifact/download/kireevco/generic/esp-open-sdk-1.1.2-linux-x86_64.tar.gz

kireevco, 2015-07-07:
Great.

closed 2015-07-10 by smitjj